sm0svx/svxlink #223: RGR-beep?

pe1ago, 2017-01-14:
I have two svxlink systems running on Raspberry Pi's. I can't stop the beep on the SQL close.
Tested with: RGR_SOUND_DELAY=-1, #RGR_SOUND_DELAY=-1 and RGR_SOUND_DELAY=0
Is it coming from somewhere else?

sm0svx, 2017-01-14:
Both commenting out and setting to -1 works. Just tested it. Are you doing the changes in the correct logic core configuration section (RepeaterLogic or SimplexLogic)?

pe1ago, 2017-01-14:
Like always: Looking al settings for weeks, google for weeks noting found. Than you ask the question en find the answer yourself looking for a other problem.

Audio beep during TX - This appears to be the same issue outlined here: http://svxlink.996268.n3.nabble.com/Beeps-in-received-audio-td3088.html "It’s a new feature which was newly added. A beep is added when a remote EchoLink transmission stops. You probably are experiencing dropped UDP packets from the remote station. If SvxLink does not receive audio for 200ms it will interpret this as if the remote transmission has ended." The fix for this is to do a local overlay replacement for the is_receiving proc in the EchoLink Tcl module in /usr/share/svxlink/events.d

    variable last_carrier_loss 0;

    #
    # Executed when a transmission from an EchoLink station is starting
    # or stopping
    #
    proc is_receiving {rx} {
       variable last_carrier_loss;

       if {$rx == 0} {
         set now [clock clicks -milliseconds];
         if { ($now - $last_carrier_loss) > 2500 } {
           playSilence 100;
           playTone 1500 100 100;
           playTone 1000 100 100;
           playSilence 100;
         }

         set last_carrier_loss $now;
       }
    }
Thanks anyway.

sm0svx, 2017-01-14:
Ok. The configuration variable you asked about only affect the local roger beep that is sent out on the systems' own transmitters.

closed 2017-01-14 by sm0svx