sm0svx/svxlink #228: ModuleFrn TX Hangtime too long

dh9yap, 2017-01-15:
FRN modules the TX Hangtime on the Freenet radio side is too long with 5-6 seconds. I tried to change this but could not find it.
It would be good if one could adjust in the FrnModule.conf.

gamecompiler, 2017-02-16:
Somebody found a solution? I think theres a problem with QSO - included into this module

sm0svx, 2017-02-17:
@sh123?

dh9yap, 2017-02-18:
We would look forward to a solution ... the module is not so usable! Hangtime on FRN 6 seconds!

sh123, 2017-02-18:
What is TX hangtime?

gamecompiler, 2017-02-18:
When we spoke into the Gateway, and the Squelch was closed, the audio transmitted contains 6 seconds air. Thats quite annoying for other room users while communication over FRN.

sh123, 2017-02-18:
It is not configurable in the module, FRN spec is asking to notify server on client TX completion not when squelch is closed, but after last voice data is received, so FRN server is notified in qso::flushSamples() with TX0 command. What is the TX hangtime for the stock FRN client?

gamecompiler, 2017-02-18:
About one second. Im not sure if id understood your question correctly (Can you speak german Oo?)
Usually when you spoke over FRN, and the voice has been transmitted, after one second - after speech - the tx line hangs up (Using the stock PC FRN client). Couldn´t it be made variable in the ModuleFrn.conf?
That would be an very nice to have feature!

Thank you very much for your attention with this problem, thats very nice. (I hope my text is grammatically correctly, if not - i just want to thank you)

sh123, 2017-02-19:
I'll test it on my machine, did not use for a while. TX hangtime is not configurable in the module, because FRN server is notified immediately without any delay when client detects that last samples are flushed. Maybe this delay comes from qso::flushSamples() called with delay or not called for some reason.

sh123, 2017-02-19:
Tried with FRN parrot "Test" channel, there is a delay about 6 seconds between squelch close and time when parrot starts sending back voice data:
```
... cut ...
req:   TX1
req:   TX1
req:   TX0   <---- (A) client indicates that it stopped transmitting
Rx1: The squelch is CLOSED (3.38704)
Tx1: Turning the transmitter ON
Tx1: Turning the transmitter OFF
cmd:   0   <--- (B) server pings back
req:   P
cmd:   0
req:   P
cmd:   0
req:   P
cmd:   0
req:   P
cmd:   2  <---- (C) server starts sending recording 
state: RX_AUDIO
voice started: <S>0</S><M>0</M><NN>CyberZoo</NN><CT>Server</CT><BC>Parrot</BC><CL>2</CL><ON>Audio Test</ON><ID>127.0.0.1</ID><DS>Audio test recorder</DS>
state: IDLE
req:   P
Tx1: Turning the transmitter ON
cmd:   2
state: RX_AUDIO
state: IDLE
req:   P
cmd:   2
state: RX_AUDIO
... cut ...
```
Delay between (A) and (C) is about 6 seconds, but client cannot do anything to decrease this delay, because that's controlled by the sever. Client only streams the voice and sends command to the server indicating when it stopped streaming.

dh9yap, 2017-02-19:
Why does it work with the original FRN customer? There is no 6 seconds.
I think his work is good, but the module can not be used properly.
Then I understood them correctly that there is no solution. From?
73 55 Jürgen :-)

gamecompiler, 2017-02-19:
sh123s sentence was a little bit confusing. But if id understood him correctly - sh123 just wanted to say that the client has to send sth like a "stop bit" to the server to terminate transmitting.

sh123, 2017-02-20:
@gamecompiler, yes, it sends tx stop command. This is the only thing client needs to do according to the FRN specification when no more voice data is available. If somebody can generate network dump with ex. wireshark from stock FRN client it might help, so it will be possible to compare.

gamecompiler, 2017-02-20:
Did a network dump while speaking and logging in. Will change my FRN password now XD (Its clearly transmitted)

[frnwireshark.zip](https://github.com/sm0svx/svxlink/files/788055/frnwireshark.zip)

sh123, 2017-02-20:
@gamecompiler , thank you for the dump, seems that stock client, does not send TX0 on stream completion, but sends RX0 instead, I'll change it and test to see if it helps.

sh123, 2017-02-20:
Made change and tested, no delays, immediate response from parrot, pull request #247

sm0svx, 2017-02-20:
Thanks! Merged.

closed 2017-02-20 by sm0svx

sh123, 2017-02-22:
@gamecompiler , @dh9yap, did you have a chance to test it with fresh changes, works OK now?

gamecompiler, 2017-02-22:
Yes it works perfectly! My gateway is now 24/7 online. Thank you very much for your fix ^^
You can speak to dh9yap and me at europa-welle.de at this moment if you want. Your are welcome.
Youd helped us a lot!

Greetings from germany,
gamecompiler