sm0svx/svxlink #236: svxlink/metarinfo

Davide75fe, 2017-02-02:
hello dear amicoTobias sv0svx, I am a user of your wonderful svxlink software I use now for 4 years with my EchoLink node in Italy, I present I IW4ENE david, I ask for a information regarding a form to enable you to say ilvero say not running for activation because servers seems that nn work, I'm talking about the "METARINFO" module which in theory would have to be activated or at least make it work for the weather speech, you know I give qlk help and care at least on how to fix it? I would keep it to work this form if we so with your help .... I await information

David
IW4ENE-L

dl1hrc, 2017-02-08:
Hi, did you have installed the latest version of SvxLink (trunk) an reconfigured your ModuleMetarInfo.conf?
There were some changes need because of termination of metar service provided by NOAA.
73s Adi / DL1HRC

Davide75fe, 2017-02-08:
Salve...premetto che io ho una versione svxlink 1.3.0,il modulo metar.info nn l ho configurato anke perche nn saprei cosa scrivere per compilazione,lei mi dice problemi per reperire i dati da Noaa non funziona?attendo risposte se può dare un aiuto e assistenza eventualmente su mio svxlink con piacere.

sm3sgp, 2017-03-10:
Hi @Davide75fe , If you are running Svxlink version 1.3.0, I think you need to upgrade. As Adi wrote, there is a fix in the latest master branch in git. If possible try that. I don't find any specific instruction on how to do that on the wiki, but the instruction about "Installation from source" is a good start. OK?

sm3sgp, 2019-09-10:
Module Metar Information is reported working fine in the latest release.

closed 2019-09-10 by sm3sgp