sm0svx/svxlink #237: ASCII serial interface

OE5FML, 2017-02-05:
Hello,

at the moment I use Echolink with Windows but I like to change to svxlink. The interface between the computer and radio is a LC-Link from Landolt (http://www.landolt.de/software/echolink.zip). This interface control the RX with VOX, the TX with ASCII serial and the DTMF is decoded external. I changed a lot of parameters in svxlink.conf but the interface didn´t work. In the manual is also no information about this interface. What should I do to make this device work?

Thank you!

73 de Michael OE5FML

dl1hrc, 2017-02-12:
Hi,
you provide a bit less information here. What parameters have you configured in the RX and TX section? Does nothing work, not even a vox? 
An ASCII TX is supported by sending 'T' an 'R' as characters over a pty device. The better way (my opinion) would be to use the RTS-pin (example 'RTS7' in the documentation) and/or 'DTR4'.
PTT_TYPE=SerialPin
PTT_PIN=DTRRTS
PTT_PORT=/dev/ttyS0 (set here the right serial port)

DTMF detection can be used in SvxLink by setting the parameters
DTMF_DEC_TYPE=EXTERNAL
DTMF_SERIAL=/dev/ttyS0  (set here the right serial device)

For a VOX (not optimal) define
SQL_DET=VOX
SQL_START_DELAY=0
SQL_DELAY=0
SQL_HANGTIME=1300

in the [RX]-section as a first approach. The Vox timing params should be adapted to your needs. Be sure that you really use the right serial port and audio channel.

73s Adi / DL1HRC

OE5FML, 2017-02-12:
Hi Adi,

thank you for the response. I use the same configuration for VOX as you wrote and this section works. 

The DTMF section is similar but the interface is another: DTMF_SERIAL=/dev/ttyUSB0 I checked the serial interface and it works, but the external DTMF decoder doesn´t work.

The PTT is also enabled with an ASCII command. What should I set at PTT_PIN to send an ACSII command? How does the pty interface work?

Thank you!

73 de Michael OE5FML

dl1hrc, 2017-02-12:
Hi Michael,

I had a mistake in the example above, for the dtmf type try DTMF_DEC_TYPE=S54S
No idea if this helps because the port settings are hard encoded in the source (9600Baud, 8N1) and there are no further information in the Landolt manual.

For the PTT I think it's better to use a line (RTS,DTR), it's supported by this interface and the easiest way. The pty is a pseudo device, not a real serial port. For the pty set:
PTT_TYPE=PTY
PTT_TYPE=/tmp/pty
You have to redirect the pty output to a serial port but no idea if this work since the port params (baudrate, bits,...) are not described in the manual and SvxLink only sends T's and R's over it.
I wrote a perl script for a NHRCx repeater controller a couple of years ago working on this principle, it's part of SvxLink. Look into svxlink/src/svxlink/scripts directory.

Why you want to use this inconvenient way, just use a single line like RTS or/and DTR.

vy 73s de Adi / DL1HRC

OE5FML, 2017-02-12:
Hi Adi,

I also tried S54S but the DTMF decoder sends no information. I think in svxlink is any bug. Echolink works very fine with the Landolt interface. At the moment I use the internal DTMF decoder. 

In the Landolt interface I changed a jumper and now the PTT works with RTS.

The svxlink runs with you help but the Landolt interface is not fully supported.

Thank you!

73 de Michael OE5FML

dl1hrc, 2017-02-12:
Hi Michael,

as I said before, I've no idea if it working this way. Since the S54S decoder has been developed for an other interface it does not necessarily work with the Landolt interface. That's why it's not automatically a bug in svxlink. The reason maybe simply the baud rate. The way of transmission isn't described in the manual.

BTW: the internal dtmf decoders working very well and you have some parameters you can change to adapt it to your hardware. So I see no disadvantage here if you use the internal decoder.
Anyway, I've built about seven repeaters and made my best experiences with very simple self-made interfaces (npn transistor, some R's and C's).
vy 73s de Adi / DL1HRC

OE5FML, 2017-02-13:
Hi Adi,

I write a message to Landolt and ask for the protocol of the interface. I will inform you! Here is the protocol of the S54S: http://lea.hamradio.si/~s57nan/ham_radio/svx_intf/svx_intf.pdf The baud rate is always a 9600 8N1 and not changeable in Echolink and svxlink.

73 de Michael OE5FML

sm0svx, 2017-02-13:
Forget about the DTMF decoder in the interface and give the software decoder a try. If audio is properly adjusted it should be as good as or better than the hardware decoder.

OE5FML, 2017-02-18:
Hello,

I ask for documentation about the Landolt interface. The developer don´t want to give any information about the ASCII protocol. Thus, the problem cannot be solved and so the internal DTMF decode have to be used.

73 de Michael OE5FML

closed 2017-02-20 by sm0svx