sm0svx/svxlink #241: SvxServer extension/features

dl1hrc, 2017-02-08:
Hi Tobias,
the svxserver, you kindly enclosed into contrib directory, is used by some hams here and partly outside of DL (e.g. SP).
The idea was to have the functionality to connect independent nodes/repeaters together. It's working without having a directory service like EchoLink. The protocol used for it is the NetTrxMsg protocol used to connect svxlink and remotetrxs. For the "normal" functionality the TCP messages implemented are sufficient but there are some information missing when this protocol is used by svxserver.

1) We have running a system of up to 20 nodes that were operated by 20 different sysops. The most of this nodes are connected via a dsl line with changeing ip addresses. So in the case of problems (open SQL) it's difficult to find out the cause to inform the right sysop about this. A good idea would be to have a tcp message that allows to send a callsign string over the protocol

2) I can understand that you wont to inflate the protocol with messages can't meaningful used by the remotetrx's but what do you think about implementing an "universal" message (int, char[255]) that can be send between svxlink and remotetrx||svxserver, e.g.

echo "abcd|123|.." > /dev/pty123 -> svxlink -> svxserver||remotetrx -> /dev/pty456

This information can be used user specific for the remotetrx's (to control relays, displays, preamplifiers, antennas, ..) on the remote site and interchange information between the connected nodes and the svxserver. A new approach is to have the state of all connected nodes available (with position-, range-, power-,... information) to inform a mobile station at the range of coverage about available repeaters/links in the vicinity.
73s de Adi/DL1HRC

sm0svx, 2017-02-11:
That is still a workaround, trying to cram the multi SvxLink node interconnect feature into the RemoteTrx protocol. I'm working right now on another solution not involving RemoteTrx:es but instead it's using a new logic core type directly. I hope to have a demonstrator out this weekend.

sm0svx, 2017-02-17:
Did not get anything working the last weekend. I'll do some more work this weekend if I can find the time.

sm0svx, 2017-02-27:
So, a first test version of SvxReflector, as I choose to call it at the moment, is now available in the svxreflector branch. It's not good enough for "production" yet since there are some important details still missing that has to do with error handling but at least the concept can be reviewed. There probably are some serious bugs as well. I'll work on it some more this week.

I guess it does the same thing as svxserver (minus some features) but it does not use the RemoteTrx protocol at all but instead there is a new logic type, ReflectorLogic, that connect directly to the svxreflector. I looked into reusing parts of svxserver but the biggest job was coming up with a new networking protocol and figuring out how to use UDP for audio so there wasn't much I could reuse.

It's quite simple to set up and get operating. The config work as usual. Just copy the default svxreflector.conf config file from the reflector subdirectory. You can also find a new ReflectorLogic config section in the svxlink.conf.in file.

The SvxReflector application need port 5300 (or whatever is configured) for incoming traffic to be open for both TCP and UDP in the firewall. There should be no need to modify the firewall on the client side.

sm0svx, 2017-03-01:
Created a pull request (#254) for SvxReflector. Continue discussion there.