sm0svx/svxlink #242: systemd not installing

kb3vgw, 2017-02-09:
dh_install -O-DWITH_SYSTEMD=ON -O-DWITH_CONTRIB_SVXSERVER=ON -O-DWITH_CONTRIB_MODULE_REMOTE_RELAY=ON -O-DWITH_CONTRIB_MODULE_TCL_SSTV=ON -O--parallel -O--buildsystem=cmake -O--sourcedirectory=src -O--builddirectory=build
cp: cannot stat ‘debian/tmp/lib/systemd/system/svxlink.service’: No such file or directory
dh_install: cp -a debian/tmp/lib/systemd/system/svxlink.service debian/svxlink-server//lib/systemd/system/ returned exit code 1

it looks like the -DWITH_SYSTEMD=ON is not woking and --with_systemd either

kb3vgw, 2017-02-09:
dh_install -O--WITH_SYSTEMD -O-DWITH_CONTRIB_SVXSERVER=ON -O-DWITH_CONTRIB_MODULE_REMOTE_RELAY=ON -O-DWITH_CONTRIB_MODULE_TCL_SSTV=ON -O--parallel -O--buildsystem=cmake -O--sourcedirectory=src -O--builddirectory=build
cp: cannot stat ‘debian/tmp/lib/systemd/system/svxlink.service’: No such file or directory
dh_install: cp -a debian/tmp/lib/systemd/system/svxlink.service debian/svxlink-server//lib/systemd/system/ returned exit code 1

kb3vgw, 2017-02-09:
with your change to my work sytemd is broken

sm0svx, 2017-02-11:
Don't know what dh_install do but when using cmake directly it's working just fine. Try this:
```
git clone http://github.com/sm0svx/svxlink
cd svxlink/src
mkdir build
cd build
cmake -DWITH_SYSTEMD=ON ..
make
make install DESTDIR=/tmp/svxlink
tree /tmp/svxlink
```
The systemd files definitely install as expected on my system. Are you sure that you are using dh_install in the correct way? Do you have any local changes that may cause the systemd files not to install?

RichNeese, 2017-10-22:
fixed

closed 2017-10-22 by RichNeese