sm0svx/svxlink #243: Echolink Auth problem

sa2blv, 2017-02-11:
Seens latest fixes.
svxlink will complie module echolink 
but when i useing it i dosent connect to the 
svx servers. it works with previus compiled module
i tested it in a simplexlogic config 

// Peter DE SA2BLV

sm0svx, 2017-02-11:
Working just fine here. Try to start over with a clean copy of the source code.

sm0svx, 2017-02-11:
Oh, if you were talking about svxserver, I have not tested it. I have no config for it here since I don't use it myself. Ask @dl1hrc if you have problems with it.

f5vmr, 2017-02-11:
I use it, don't forget to set the passwords at both ends.

sa2blv, 2017-03-08:
the peolem sems to be in svxlink when it supouse to connect to the echolink server
i dosent get an answer from server and promt the echolink server. the build was clean new 
from git but cant make this to happen on an other system

closed 2017-03-08 by sa2blv