sm0svx/svxlink #244: Audio drops sometimes

dl1esk, 2017-02-15:
Hi there,

i`ve been using Svxlink since a while....
I have 2 Repeaters running on Svxlink on RPI 2 and both have the same issue when i set dwc_otg_speed =1
With USB 2.0 Speed there is an Audio Delay with Svxlink so that u can hear ur last spoken word sometimes.
When i switch USB to 1.1 there is no Audio Delay and everything is fine but sometimes i get an Audio drop 
When the Repeater is opened it happens one Time and then u can speak hrs and it doesnt happen again ( i think )... But when the Repeater closes and when u open it again then it happens again one time..it`s not reproducible that the Repeater doing it at the same time etc...
What i saw is that Svxlink`s Processor load dropped down for a Second and came back to normal...
Any Ideas what the problem is ?
I use GPIO for Squelch Detection and PTT...

73 Sebastian

dl1esk, 2017-02-16:
another Info..when this happens u can hear 2 or 3 letters from the spoken Word over the Repeater and the rest is without Audio as long as that person speaks at this moment.
When the person releases the PTT u hear the Roger Beep and the next Transmissions are fine

dl1esk, 2017-02-17:
I did some tests on the Repeater and could reproduce the Audio drops....
If the Squelch opens just for a few milliseconds u become an Audio drop...

DG1BBQ, 2017-02-17:
I can confirm this behavior. I set up the software configuration for the repeater DB0NGU running on a Raspberry Pi 2. A station on RF starts talking, you hear one sillable, then a short dropout and the rest of the transmission is fine, but  following transmission can display the same beahaviour sporadically.
The phenomenon is pretty much identical to what I've observed when using remotetrx to link to a repeater.

sm0svx, 2017-02-17:
The RPi unfortunately seem to be a very unreliable platform to run SvxLink on. For some people it works great and other people have all sorts of strange problems. USB seem to be very flakey on that device. I've never heard people reporting these kind of strange problems on any other hardware. I'd suggest trying another type of sound dongle, try the RPi 3 or some other single board computer. Some sound dongles seem to work better than others on the RPi.

I know the RPi is very popular since it's very easy to get hold of and very reasonably prised. Unfortunately you get what you pay for.

johnhays, 2017-02-17:
One option is the UDRC-II -- http://nwdigitalradio.com/product/udrc/

(Tobias, would you contact me off list?)

k7ve@k7ve.org

dl1esk, 2017-02-18:
Hi Tobias,

i tried different Soundcards, every Soundcard has the same Problems with Audio Delay on USB 2.0....
u can eliminate it with the USB 1.1 Setting but then u bought that Audio hangups :-)
what me confuses is that in this moment when this happens the load of svxlink drops down under the normal load....
@ John i cant use that Board because i need my GPIO pins because i use them to read an A/D controller to read the RSSI from RX to make Signal strengh dependent Roger Beeps :-)

johnhays, 2017-02-18:
If you can pick which GPIO pins you use for rssi a/d there are left over
GPIO pins and pass through pads on the UDRC board.

dl1esk, 2017-02-18:
@John i use SPI for the AD Controller and i think u also use that Pins :-)
I don`t think that this is an USB Issue because it happens one Time when the Repeater is opended and then u can Talk hrs and the error doesn`t come again thats confusing....
Atm we talk over the Repeater since 2hrs it happened in the beginning one time and now we can talk without any issues...

johnhays, 2017-02-18:
@dl1esk - Yes, you are correct, the UDRC uses the SPI for the CODEC.

johnhays, 2017-02-19:
Updated information from the design engineer

The UDRC does NOT use SPI for control of the CODEC, it uses I2C.

On Feb 18, 2017, at 3:35 PM, John D. Hays <john@hays.org> wrote:

@dl1esk - Yes, you are correct, the UDRC uses the SPI for the CODEC.

On Sat, Feb 18, 2017 at 2:12 PM, dl1esk <notifications@github.com> wrote:

> @John <https://github.com/John> i use SPI for the AD Controller and i
> think u also use that Pins :-)
> I dont think that this is an USB Issue because it happens one Time when
> the Repeater is opended [...]
------------------------------
John D. Hays
K7VE

PO Box 1223, Edmonds, WA 98020-1223
   <http://k7ve.org/blog>  <http://twitter.com/#!/john_hays>

kb3vgw, 2017-04-04:
look at getting a fe-pi and move away from usb sound cards. also have a look at the svxlink cards in development we have a few that are out there and the 2 new ones due out by hamvention will have codecs onboard so no more usb sound cards.

sm3sgp, 2019-09-10:
@dl1esk @DG1BBQ Seastian, do you still see this problem?
I sometimes hear audio drops, but then its is related to high cpu load or network load.

closed 2025-03-18 by sm3sgp