sm0svx/svxlink #245: Audio clips do load at all

ajkatz83, 2017-02-15:
When playing the short ID, I get warning cannot find phonetic letter in context or whatever. the TCL system is not very comprehensive at all and I have no clue how to get these audio files to process right seeing the whole event handler system is python or C or something which makes it harder. the sounds are in the default directory but never are found when svxlink goes to ID. I just dunno what to think anymore.

ajkatz83, 2017-02-15:
They do not load

sm3sgp, 2017-02-16:
How did you install svxlink? If you installed it by downloading the source and compile yourself, you also need to download and install the sound files manually.

Maybe you have already done that, but this is what is written in the manual page:
_If you are going to run the SvxLink server, unpack the sound files at a good location. The normal location for sound files is /usr/share/svxlink/sounds/. As user root, run the commands below. Adapt the commands to your needs by specifying the correct language code, name and version._

_cd /usr/share/svxlink/sounds
tar xvzf /path-to-wherever-you-put-the-tar-file/svxlink-sounds-ll_LL-name-16k-YY.MM.tar.bz2
ln -s ll_LL-name-16k ll_LL_

The last part with the link is important.  It will give you something like this:
[sm3sgp@sture sounds]$ pwd
/usr/share/svxlink/sounds
[sm3sgp@sture sounds]$ ls -l
total 8
lrwxrwxrwx.  1 root root    17 Jul 20  2014 en_US -> en_US-heather-16k
drwxr-xr-x. 12 1000 users 4096 Dec  1  2013 en_US-heather-16k
lrwxrwxrwx.  1 root root    14 Aug  8  2014 sv_SE -> sv_SE-elin-16k
drwxr-xr-x.  9 1000 users 4096 Aug  8  2014 sv_SE-elin-16k

In the your Logic section in svxlink.conf you should be able to point to your language pack, if not specified it will default to: DEFAULT_LANG=en_US

closed 2017-03-06 by ajkatz83