sm0svx/svxlink #250: SVXLink and Asterisk ???

sp2ong, 2017-02-22:
What about possibility connect svxlink with asterisk PBX VoIP ??
We have a lot of running asterisk PBX in Europe

https://www.afu.rwth-aachen.de/dundicrawler/

it will be nice to have possibility connect to local echolink node from VoIP hardware phone. And from echolink node use DTMF can connect to VoIP phone.

I have found on video which show this possibility:

https://www.youtube.com/watch?v=ym4WwEuQYxk

Anybody know how to configure svxlink / svxserver and asterisk to do this ???

dl1hrc, 2017-02-22:
> What about possibility connect svxlink with asterisk PBX VoIP ??

working on it...

73s de Adi / DL1HRC

sp2ong, 2017-02-22:
Ok good news :-) 

73 Waldek SP2ONG

dl1hrc, 2017-02-23:
Am 22.02.2017 um 19:42 schrieb Waldek:
> Ok good news :-)
> 
> 73 Waldek SP2ONG
> 

Hello Waldek,

first tests were sucessful. It's in a very early state so some errors and problems may still occur!
Here a very short howto for people who know what they are doing.
Will try to make a documentation (at first in German on http://svxlink.de) when I find the time.

You need to have asterisk and SvxLink running on the same hardware. Audio will be routed through alsa loopback devices.
With "modprobe snd-aloop" you get up to 8 alsa lb devices:

Karte 2: Loopback [Loopback], Gerät 0: Loopback PCM [Loopback PCM]
  Sub-Geräte: 7/8
  Sub-Gerät #0: subdevice #0
  Sub-Gerät #1: subdevice #1
  Sub-Gerät #2: subdevice #2
  Sub-Gerät #3: subdevice #3
  Sub-Gerät #4: subdevice #4
  Sub-Gerät #5: subdevice #5
  Sub-Gerät #6: subdevice #6
  Sub-Gerät #7: subdevice #7
Karte 2: Loopback [Loopback], Gerät 1: Loopback PCM [Loopback PCM]
  Sub-Geräte: 8/8
  Sub-Gerät #0: subdevice #0
  Sub-Gerät #1: subdevice #1
  Sub-Gerät #2: subdevice #2
  Sub-Gerät #3: subdevice #3
  Sub-Gerät #4: subdevice #4
  Sub-Gerät #5: subdevice #5
  Sub-Gerät #6: subdevice #6
  Sub-Gerät #7: subdevice #7

in Asterisk:

Look for asterisk "Console" configuration. If you compile it, then enable support for alsa and portaudio.
In asterisk/alsa.conf define the audio ports pointing to alsa loopback devices:

input_device=hw:2,0,0
output_device=hw:2,0,1

in asterisk/extensions.conf you need a reference to call the console when an incoming call is received:

exten => 2###0,1,Dial(Console/default)
exten => 2###0,n,Hangup

SvxLink:

Define an own Logic for Asterisk as a repeater, e.g.:

[AsteriskLogic]
TYPE=Repeater
RX=RxAsterisk
TX=TxAsterisk
OPEN_ON_SQL=100
OPEN_ON_DTMF=*
DEFAULT_LANG=de_DE
OPEN_SQL_FLANK=OPEN
IDLE_TIMEOUT=10
NO_REPEAT=1
CALLSIGN=DB0HRC-A
EVENT_HANDLER=/usr/share/svxlink/events.tcl
DEFAULT_LANG=de_DE

[RxAsterisk]
TYPE=Local
AUDIO_DEV=alsa:plughw:2,1,1
AUDIO_CHANNEL=0
SQL_DET=VOX
SQL_START_DELAY=0
SQL_DELAY=0
SQL_HANGTIME=1300
VOX_FILTER_DEPTH=20
VOX_THRESH=1000
DEEMPHASIS=0
SQL_TAIL_ELIM=0
DTMF_DEC_TYPE=INTERNAL
DTMF_MUTING=1
DTMF_HANGTIME=100
DTMF_SERIAL=/dev/ttyS0
DTMF_MAX_FWD_TWIST=18
DTMF_MAX_REV_TWIST=12

[TxAsterisk]
TYPE=Local
AUDIO_DEV=alsa:plughw:2,1,0
AUDIO_CHANNEL=0
PTT_TYPE=NONE
TIMEOUT=9000
TX_DELAY=0
PREEMPHASIS=0

Connect the AsteriskLogic and your RepeaterLogic by LogicLinking, remember there is no security at the moment (e.g. PIN-query).

Good luck, vy 73s de Adi / DL1HRC

sp2ong, 2017-02-23:
Hi Adi,

many thanks for info and no problem to for me to move svxlink to physical one server.
I have asterisk v13.x own compiled and not problem to recompile with alsa and portaudio support

I will try to do according to your suggestion

73 Waldek SP2ONG

dl1hrc, 2017-02-23:
Hi Waldek,

it was my first contact with asterisk two days ago, so I'm an absolute beginner here and very interested in having further information on pin queries, incoming call handling, e.g. call to 555-1234-001 connects to DB0HRC, 555-1234-002 to DB0HAL and so on.
Please let me know what you find out, can also be that alsa loopback's are not the best way.

vy 73s de Adi / DL1HRC

sp2ong, 2017-02-23:
Ok I will be contact with you when I make progress with my configuration or find problems

73 Waldek SP2ONG

sp2ong, 2017-02-24:
Hi Adi,

Your proposal is nice when we have possibility move/run echolink on one PC but in many cases echolink nodes are running on small terminal and mounted very close to radio installation. But asterisk PBX in many cases are part of HamNET infrastructure and installed on more powerful CPU server for asterisk for this reason we need still second solution to have possibility connect echonlink node via network with asterisk. Or write small addon to echolink similar svxserver 'asterlink' use IAX technology and OPUS codec to connect with asterisk or write addon to asterisk similar like Allstar Link. The Allstar link unfortunately base on old version asterisk 1.x we are use news version asterisk 13.x or 14.x .

I would like thank you  that you have find solution  for echolink and asterisk installed on one PC but maybe in future we find solution connect via network echolink with asterisk

In this weekend I will be try install svxlink on my Asterisk PBX server to see how this solution working

73 Waldek SP2ONG

sm0svx, 2017-02-24:
You could do a setup where you run a RemoteTrx on the Asterisk system and the main SvxLink system somewhere completely different. On the RemoteTrx side use the same rx and tx config as Adi suggest.

dl1hrc, 2017-02-25:
Yes, I think it's a good solution to do it this way, since the native implementation of protocols like IAX2, SIP, ... into SvxLink needs a lot of external libraries and of course much more manpower in SvxLink development.
At the moment I'm looking for a solution to connect the alsa device to a chat room (bridge) while asterisk is starting up. A pin queue (to access the chat room from the phone line) is the condition to a asterisk<->SvxLink interconnect.
73s de Adi / DL1HRC

sp2ong, 2017-02-25:
Hi,

Adi, connected svxlink to conference bridge will be nice but I think only allow speak with other who connected to conference root from asterisk users.

But I think maybe not now but in near future, we have possibility interconnect svxlink with asterisk. When I will be use only radio and connect to echolink node which have interconnect with asterisk I can send via DTMF for example your phone VoIP number 315310427323 ( if I know that you have and use asterisk https://www.oe2wnl.at/calltodtmf-voip.php?call=DL1HRC&submit=Convert ) and I will be can speak with you from may handy radio via echolink node i Poland Torun with you, where you will be speak to your hardware/software VoIP asterik phone in Germany :-)
Our asterisk PBX in HamNEt Europe use dundi to interconnect with other asterisk PBX for this reason I can select voip phone number on my local asterisk to any hamradio operator in HamNET Europe to have possibility speak with other hams in Europe.
We use 2 different voip technology like echolink and asterisk (SIP/IAX) and only we need merge these.
I have wrote that mayb we need look to Allstar sources to find idea or other to do this because Allstar is hybrid base on asterisk where exist asterisk SIP/IAX, Allstar node and have possibility configure as echolink node. But one problem with Allstart that use very old asterisk 1.x and don't exist app_rpt module to new version asterisk   

I know that it is not easy task to do.

73 Waldek SP2ONG

sp2ong, 2017-02-25:
I have found app_rpt sources in svn allstar link:

http://svn.ohnosec.org/svn/projects/allstar/astsrc-1.4.23-pre/trunk/asterisk/apps/app_rpt.c

sp2ong, 2017-02-26:
Hi Adi,

I have try to do following with your instruction on:

http://svxlink.de/?page_id=3411

but I have following errors:
===============================================================
SvxLink v1.5.99.5 Copyright (C) 2003-2017 Tobias Blomberg / SM0SVX

SvxLink comes with ABSOLUTELY NO WARRANTY. This is free software, and you are
welcome to redistribute it in accordance with the terms and conditions in the
GNU GPL (General Public License) version 2 or later.

Using configuration file: /etc/svxlink/svxlink.conf

Starting logic: SimplexLogic
Loading RX: Rx1
Loading TX: Tx1
Loading module "ModuleHelp" into logic "SimplexLogic"
	Found /usr/lib/i386-linux-gnu/svxlink/ModuleHelp.so
	Module Help v1.0.0 starting...
Loading module "ModuleParrot" into logic "SimplexLogic"
	Found /usr/lib/i386-linux-gnu/svxlink/ModuleParrot.so
	Module Parrot v1.1.1 starting...
Loading module "ModuleEchoLink" into logic "SimplexLogic"
	Found /usr/lib/i386-linux-gnu/svxlink/ModuleEchoLink.so
	Module EchoLink v1.3.99.0 starting...
SimplexLogic: Event handler script successfully loaded.

Starting logic: RepeaterLogic
Loading RX: Rx1
Loading TX: Tx1
Loading module "ModuleHelp" into logic "RepeaterLogic"
	Found /usr/lib/i386-linux-gnu/svxlink/ModuleHelp.so
	Module Help v1.0.0 starting...
Loading module "ModuleParrot" into logic "RepeaterLogic"
	Found /usr/lib/i386-linux-gnu/svxlink/ModuleParrot.so
	Module Parrot v1.1.1 starting...
Loading module "ModuleEchoLink" into logic "RepeaterLogic"
	Found /usr/lib/i386-linux-gnu/svxlink/ModuleEchoLink.so
	Module EchoLink v1.3.99.0 starting...
RepeaterLogic: Event handler script successfully loaded.

Starting logic: AsteriskLogic
Loading RX: RxAsterisk
Loading TX: TxAsterisk
AsteriskLogic: Event handler script successfully loaded.
*** ERROR: Unable to handle event: AsteriskLogic::startup in logic AsteriskLogic (invalid command name "AsteriskLogic::startup")
Activating link AsteriskRepeaterLink
EchoLink directory status changed to ON
EchoLink directory status changed to ON
--- EchoLink directory server message: ---
EchoLink Server v2.5.9997

ECHO6: N.Virginia, USA
======================================

My base configuration for this test before changes was:
[GLOBAL]
LOGICS=SimplexLogic
CFG_DIR=svxlink.d
TIMESTAMP_FORMAT="%c"

I have add all according your describe in svxlink.conf with change GLOBAL follow:
[GLOBAL]
LOGICS=SimplexLogic,RepeaterLogic,AsteriskLogic
CFG_DIR=svxlink.d
TIMESTAMP_FORMAT="%c"
LINKS=AsteriskRepeaterLink

dl1hrc, 2017-02-26:
yes, forgot to tell you that you need a new Tcl-Namespace "AsteriskLogic"

cp /usr/share/svxlink/events.d/RepeaterLogic.tcl 
/usr/share/svxlink/events.d/AsteriskLogic.tcl

and change the line at the beginning of the file from

namespace eval RepeaterLogic {

to

namespace eval AsteriskLogic {

vy 73s de Adi

http://svxlink.de/?page_id=3411 > > > but I have following errors: > > 
SvxLink v1.5.99.5 Copyright (C) 2003-2017 Tobias Blomberg / SM0SVX > > 
SvxLink comes with ABSOLUTELY NO WARRANTY. This is free software, and > 
you are welcome to redistribute it in accordance with the terms and > 
conditions in the GNU GPL (General Public License) version 2 or > later. 
 > > Using configuration file: /etc/svxlink/svxlink.conf > > Starting 
logic: SimplexLogic Loading RX: Rx1 Loading TX: Tx1 Loading > module 
"ModuleHelp" into logic "SimplexLogic" Found > 
/usr/lib/i386-linux-gnu/svxlink/ModuleHelp.so Module Help v1.0.0 > 
starting... Loading module "ModuleParrot" into logic "SimplexLogic" > 
Found /usr/lib/i386-linux-gnu/svxlink/ModuleParrot.so Module Parrot > 
v1.1.1 starting... Loading module "ModuleEchoLink" into logic > 
"SimplexLogic" Found > /usr/lib/i386-linux-gnu/svxlink/ModuleEchoLink.so 
Module EchoLink > v1.3.99.0 starting... SimplexLogic: Event handler 
script successfully > loaded. > > Starting logic: RepeaterLogic Loading 
RX: Rx1 Loading TX: Tx1 Loading > module "ModuleHelp" into logic 
"RepeaterLogic" Found > /usr/lib/i386-linux-gnu/svxlink/ModuleHelp.so 
Module Help v1.0.0 > starting... Loading module "ModuleParrot" into 
logic "RepeaterLogic" > Found 
/usr/lib/i386-linux-gnu/svxlink/ModuleParrot.so Module Parrot > v1.1.1 
starting... Loading module "ModuleEchoLink" into logic > "RepeaterLogic" 
Found > /usr/lib/i386-linux-gnu/svxlink/ModuleEchoLink.so Module 
EchoLink > v1.3.99.0 starting... RepeaterLogic: Event handler script > 
successfully loaded. > > Starting logic: AsteriskLogic Loading RX: 
RxAsterisk Loading TX: > TxAsterisk AsteriskLogic: Event handler script 
successfully loaded. > *** ERROR: Unable to handle event: 
AsteriskLogic::startup in logic > AsteriskLogic (invalid command name 
"AsteriskLogic::startup") > Activating link AsteriskRepeaterLink 
EchoLink directory status > changed to ON EchoLink directory status 
changed to ON --- EchoLink > directory server message: --- EchoLink 
Server v2.5.9997 > > > ECHO6: N.Virginia, USA > > My base configuration 
for this test before changes was: [GLOBAL] > LOGICS=SimplexLogic 
CFG_DIR=svxlink.d TIMESTAMP_FORMAT="%c" > > I have add all according 
your describe in svxlink.conf with change > GLOBAL follow: [GLOBAL] > 
LOGICS=SimplexLogic,RepeaterLogic,AsteriskLogic CFG_DIR=svxlink.d > 
TIMESTAMP_FORMAT="%c" LINKS=AsteriskRepeaterLink > > — You are receiving 
this because you commented. Reply to this email > directly, view it on 
GitHub > 
<https://github.com/sm0svx/svxlink/issues/250#issuecomment-282549588>, > 
or mute the thread > 
<https://github.com/
 >

sp2ong, 2017-02-26:
Hi Adi,

Ok, thanks now svxlink running without errors. Next,  I will be try tune configuration asterisk

73 Waldek SP2ONG

dl1hrc, 2017-02-26:
There ist one remaining thing:

Since SvxLink is running with 48k sampling rate (normally) and Asterisk 
expect 8k, you have to start Asterisk first, then SvxLink. Otherwise you 
will get an error message from Asterisk.

The better way would be to have an Alsa samplerate converter in the 
asound.conf.

vy 73s de Adi / DL1HRC

sp2ong, 2017-02-26:
Ok Adi but I have small problem with asterisk , when I add and changed all necessary files for asterisk and restart asterisk in log I have fond following message

[Feb 26 13:12:40] WARNING[1501] channel.c: Already have a handler for type 'Console'
[Feb 26 13:12:40] ERROR[1501] chan_alsa.c: Unable to register channel class 'Console'

and asterisk  not running
It is problem when is loading chan_console.so

Do you have example asound.conf to convert sampling rate ?

dl1hrc, 2017-02-26:
Same message her, no idea what's the reason is for it. But it's working 
well here. I think we have to google for a solution.

BTW: There are a lot of notice messages here since I use the default 
configuration.

I do not have a asound.conf for now, my test configs were creating 
errors. Will look later when I have more time :)

sp2ong, 2017-02-26:
Ok I will ignore this but can you explain more about

exten => 2####0,1,Authenticate(1234)
exten => 2####0,2,Playback(access-granted)
exten => 2####0,3,Dial(Console/default)
exten => 2####0,n,Hangup()

I am not sure what I need add to [default] 

My extension.conf 

[general]
static=yes
autofallthrough=yes
writeprotect=no
clearglobalvars=no
autofallthrough=yes

[default]
exten => _X.,1,Hangup()

[dundiextens]
exten => _7471206000,1,Goto(conference,${EXTEN},1)

[conference]
exten => 7471206000,1,Noop()
same => n,ConfBridge(room-one,default_bridge,default_user,sample_user_menu)

[lookupdundi]
switch => DUNDi/priv

[local]
include => dundiextens

exten => _Z.,1,Dial(SIP/${EXTEN},30)
same => n,Voicemail(${EXTEN}@voicemail)
same => n,Goto(lookupdundi,${EXTEN},1)
same => n,Hangup()
             
exten => 7471206000,1,Noop()
same => n,ConfBridge(room-one,default_bridge,default_user,sample_user_menu)

[incomingdundi]
exten => _Z.,1,Goto(local,${EXTEN},1)

dl1hrc, 2017-02-26:
puuhhh, I'm still a beginner in asterisk ;)
You need a call handler like this, where 2####0 stands for your sip number (e.g. 234580)

exten => 2####0,1,Authenticate(1234)
exten => 2####0,2,Playback(access-granted)
exten => 2####0,3,Dial(Console/default)
exten => 2####0,n,Hangup()

At the moment it's not possible to connect the Console to a conference because Dial(Console/default) does not return the connection handler the to be connected to the ConfBridge(xx,yy,zz)
I recommend to do it first as it explained in my example to be sure that it's working well. Then we can looking for better solutions.

sp2ong, 2017-02-26:
Ok Adi,

If I am understood that any number 2xxxx0 incoming from console for example

212340
or
234230
etc

will be pass to Hangup
because we use phone number converted from callsign to DTMF code and are long for exmaple my number is 747120636241
I need make aliases shorter to use via DTMF from radio channel

I am not sure that I need this

exten => 2####0,1,Authenticate(1234)
exten => 2####0,2,Playback(access-granted)
exten => 2####0,3,Dial(Console/default)
exten => 2####0,n,Hangup()

move to [local] part extensions.conf

It is look that I must read documentation asterisk about extensions.conf

73 Waldek SP2ONG

dl1hrc, 2017-02-26:
The function I wanted to have first was a simple way to dial-in into hamradio repeater-network from outside (not hamnet). It's just a beginning, asterisk has almost infinite functions and facilities to connect other networks. 
So have fun to extend the functionality, maybe we will have a direct connection (IAX,...) to asterisk later.

vy 73s de Adi / DL1HRC

sp2ong, 2017-02-26:
Adi,

I have try understand how this all working and read about alsa loop. I have found example

http://www.getreu.net/public/downloads/doc/snd-aloop-device/snd-aloop-device.html#connect

and it is  look like in asla.conf we need ??? following :

input_device=hw:2,0,0 
output_device=hw:2,1,0

I thing in my case extensions.conf I don't need dial plane with 2####0 
I have similar configuration like use Frank DL3DCW in hamserverpi
but still don't have success with calling VoIP number from echolink node.
 
73 Waldek SP2ONG

dl1hrc, 2017-02-26:
the reason may be the LogicLinking, do you use DTMF_MUTING=1 in your 
RX-configuration?

You can try a different configuration with a Voter and MultiTx too

sp2ong, 2017-02-28:
Hi Adi,

I don't use DTM_MUTING=1 in my RX configuration. I have still problem maybe with pass DTMF tones ?
Because for tests I have run on server where I have asterisk run svxlink without radio access. I have connect via network use echolink software clinet and send DTMF tones from my online DTMF tones
It maybe for this reason I have problem. Configuration of svxlink is simple base on distribution svxlink only add your suggested options to have link with asterisk. Maybe when I send DTMF tones from my echolink clinet software they are not passed to asterisk.

dl1hrc, 2017-02-28:
Am 28.02.2017 um 07:00 schrieb Waldek:
> Hi Adi,
>
> I don't use DTM_MUTING=1 in my RX configuration. I have still problem maybe with pass DTMF tones ?
> Because for tests I have run on server where I have asterisk run svxlink without radio [...]

Hi Waldek,

I think it's not quite simple since EchoLink normally uses the gsm codec as a voice codec.
The tones were not clearly transmitted over this way. You should try to use the message channel for it.

sp2ong, 2017-02-28:
Adi,

I am not sure but we can send DTMF codes and work only when we connected directly via radio with echolink node which have interlink with asterisk.

Do you try test that other or you connect to your echolink node not directly via radio but from other echolink node and try make connect to asterisk use VoIP number via DTMF ??? 

73 Waldek

dl1hrc, 2017-02-28:
At the moment I do not test it to dial from RF into the phone network
since it's not needed at the moment here. I have a commercial sip
contract and for me it's to not safe enough over the air with simple DTMF.

Maybe I can test it locally next time without any RF environment, but it
will take some days to realize it.

73 Adi

VR2XKP, 2017-06-02:
That's great Svxlink worked with Asterisk. However, I would like to ask does the above configuration request svxlink & asterisk install to  the same machine (e.g. Raspberry pi 2 or 3), or 2 individual machine.

Tks,
Francis

sp2ong, 2017-06-02:
Hi Francis,

Adi described ( https://svxlink.de/?page_id=3411 )  asterisk configuration
that must be installed on the same computer on which svxlink is running.

73 Waldek

sm0svx, 2017-06-19:
@VR2XKP: As stated above, it may be possible to use a RemoteTrx if you need to have Asterisk on one computer and the main SvxLink node on another. However, I have not studied Adi's instruction in detail so there may be something that prevent it. Adi can probably clarify this.

dl1hrc, 2017-06-20:
I never tried it but I'm sure that you can use the RemoteTrx as well. Since the alsa loopback devices are used either SvxLink or RemoteTrx must be run on the same engine.

sp2ong, 2017-06-20:
Wow, nice idea with integration Asterisk on another server - that was my problem where I have asterisk HamNET is running on another server than echolink node. In remotetrx.conf we need only define
RxAsterisk and TxAsterisk with loopback sounds for NetUplinkTrx to interact with Asterisk

dl1hrc, 2017-06-20:
Waldek, please could you give me a short feedback when it works as expected? Thank you.

sp2ong, 2017-06-21:
I have run remotetrx on server Asterisk and link with svxlink server. I have recorded sound file from loopback device defined in asterisk alsa.conf 

input_device=hw:2,0,0

use following command:

 arecord -r  16000 -f S16_LE -D hw:2,0,0 test.wav

I have play DTMF number via radio and it looks that DTMF numbers were recorded in test.wav file for this reason I know that DTMF is passed from svxlink server to remotrx and appear on loopback device hw:2,0,0
But asterisk doesn't decode or another problem is that I can not connect to the phone number.
I can not idea how to debug more asterisk to find a problem. I have use CLI asterisk

asterisk -r vvvvvvvv 

but I can not see any messages to help me understand a problem.
The alsa.conf and console.conf I have from your description only in the extension configuration I have put for test:

[globals]
CONSOLE=Console/default

[default]
exten =>12345,1,Dial(SIP/747120636241,30)
exten =>12345,n,Hangup()

to redirect short number 12345 to my SIP Hamnet phone number

I have loaded modules chan_console and chan_alsa and no load chan_oss 
73 Waldek

dl1hrc, 2017-06-22:
What codec do you use between RemoteTrx and SvxLink node? Could you try with RAW again?

sp2ong, 2017-06-22:
I have use OPUS but I will try RAW

sp2ong, 2017-06-23:
Hi Adi, I have switched to RAW but still is problem decode DTMF by Asterisk. 
But I need confirmation that Asterisk working correctly via alsa channel. Do you or others have console tools to play DTMF code and send to loopback audio ?? I have found simple tool with wav files

http://sp2pmk.tvk.torun.pl/uploads/dtmf.tar.gz
But when play DTMF code to dial number still asterisk don't work. For this tools need mplayer or wvaplay
https://sourceforge.net/projects/wavplay/

I have check asterisk that chan_alsa is loaded with 

chan_alsa.so                   ALSA Console Channel Driver              0          Running          extended
chan_console.so                Console Channel Driver                   0          Not Running      extended

Adi do you have an idea how to check that channel alsa asterisk work without svxlink ?

73 Waldek

sp2ong, 2017-06-23:
I have use script dmtf which play numbers

dtmf 1234

but asterisk don't decode via alsa channel input.  I have use wav file with sample rate 8kHz and 16 kHz

dl1hrc, 2017-06-23:
As far as I know asterisk is working only with a 8k stream over the alsa channel.

sp2ong, 2017-06-23:
Ok Adi I have replaced dtmf.tar.gz with wav files to 8 k sample rate if anybody would like use this tools
to send directly DTMF code to alsa channel input of asterisk to check  that will be decode and dial

sp2ong, 2017-06-23:
Adi can you try use this tools to send dtmf code to alsa channel input asterisk to check that on your configuration working ?
For play wav files via this script you must install mplayer

sp2ong, 2017-06-23:
I am not sure but maybe we need create file /etc/asound.conf with the definition of loopback device which is used in asterisk ???

 pcm_slave.hw_loopback {
   pcm "hw:Loopback,0,2"
   channels 2
   format RAW
   rate 8000
 }

 pcm.plug_loopback {
   type plug
   slave hw_loopback
     ttable {
     0.0 = 1
     0.1 = 1
   }
 }

sp2ong, 2017-06-23:
Adi,

I have found information in asterisk sources modelus.conf.sample that:
; Load one of: chan_oss, alsa, or console (portaudio).
; By default, load chan_oss only (automatically).
;
load => chan_alsa.so
noload => chan_oss.so
noload => chan_console.so

If we use ALSA channel we don't need load chan_console and define this module.

dl1hrc, 2017-06-23:
Hm, version 1.2.x? Maybe is to old?
I'm running 14.3.0
BTW: Sorry, I'm quite busy at the moment an can only answer with a high delay :/

sp2ong, 2017-06-23:
I have use asterisk v13..0.8 and when I have load only chan_alsa I don't have any erroros about channel console and from CLI I can dial via alsa for example:

console dial 1234 

It is loo that for asterisk version 1.2 and higger we need decised use only one console channel
chan_alsa , chan_oss or chan_console (which use portaudio).  

Informatin this: " Load one of: chan_oss, alsa, or console (portaudio)." is from sources v13.0.8

sp2ong, 2017-06-23:
Adi this information about load only one channel console type you can find:

https://github.com/asterisk/asterisk/blob/master/configs/samples/modules.conf.sample

dl1hrc, 2017-06-23:
Ok, thanks, will try it here.

dl1hrc, 2017-06-23:
ok, with noload => chan_console.so it's working as well :)
I think a good way would be to register at https://community.asterisk.org/ to ask for help since I have very low experiences with asterisk.

sp2ong, 2017-06-23:
I have put on extensions.conf in default following

[default]
exten => _1234,1,Dial(SIP/74712063624110,30)
exten => _1234,n,Hangup()

When I run in CLI asterisk command:

CLI> console dial 1234

asterisk connect to my SIP hamnet phone and I can under linux play DTMF code with script

dtmf 12234

and I hear this DTMF code in my SIP phone

but when I have disconect sip phone and agin play DTMC code under linux  it is look like 
asterisk don't decode DTMF sounds in my asterisk. Maybe I need upgrade my asterisk to version from 13 to 14

sp2ong, 2017-06-23:
Adi 

I have asked on Asterisk community and I have the answer: 

"The alsa channel has no built in support for looking for DTMF in the audio."

others channesl like chan_oss and chan_consloe have the same problem and:

"...  those modules expect to receive an audio stream that would contain DTMF in them, so none will do that."

It is not good news. Allstar code have own version alsa channel "chan_alsaradio" but is is code for old version asterisk
It is look at current we don't have way to send DTMF code from svxlink to asterisk with decode dtmf tones via asterisk channel

sp2ong, 2017-06-23:
I have information about chan_alsaradio 

https://yo3iiu.ro/blog/?p=361

and sources

http://www.yo3iiu.ro/archives/chan_alsaradio/chan_alsaradio.c_02.tar.gz

To correctly build the chan_alsaradio driver, the following line needs to be added in channels/Makefile:

chan_alsaradio.so: LIBS+=-lasound

But I don know that it is possible to add to current version asterisk

dl1hrc, 2017-06-23:
Hm, but maybe there is a chance to send a dtmf dial string to asterisk over another channel. There is no problem on SvxLink to put the received dmtf digits out (pty or tcl) and pipe them to a script that may initiate a dial command on asterisk. Could you find out that? Sri, I have less time this weekend :/

sp2ong, 2017-06-24:
I have tried complied chan_alsaradio 
http://www.yo3iiu.ro/archives/chan_alsaradio/chan_alsaradio.c_02.tar.gz 
with asterisk v13.8 but without success and it looks that this code is needed the rewrite to newest asterisk code because version 0.2 was created to asterisk 1.x
I think that chan_alsaradio source (of course after adoption to new asterisk code) is at current one way to have a link between svxlink and asterisk but we must find a developer who does this.

sp2ong, 2017-06-29:
I think that it maybe better solution will add module to svxlink **IAXAsterisk** to connect with asterisk via network TCP/IP 

https://www.voip-info.org/wiki/view/Asterisk+IAX+channels
https://www.voip-info.org/wiki/view/IAX

this method will be independent to actually sources asterisk version 
We can base on IAXClinet sources: https://sourceforge.net/projects/iaxclient/
and information https://www.codeproject.com/Articles/15277/Using-the-Asterisk-IAXClient-library-in-C

But maybe someone else has a better idea  how to integrate svxlink with voip asterisk HamNet network

gamecompiler, 2017-09-15:
Hi, ive got some questions:
Is the "asterisk module" included in the current masterbranch?
How do i hang up the call on the radio side?
How can i dial on the radio side?

What will happen when i lost the connection to the SIP proxy?
I would like to built two testing gateways, thats why im asking.

Thanks in advance,
gamecompile

sp2ong, 2019-04-08:
https://github.com/sm0svx/svxlink/pull/437

closed 2019-04-08 by sp2ong