sm0svx/svxlink #255: Connect issues

ajkatz83, 2017-03-06:
I know I am close with what I got going with svxlink so far, but there is a strange connection anomaly that goes on. Locally on the computer and RPi and can connect into the node find and raise Echolink. I can use the app on my phone and it shows to be there in the directory and I can connect to myself just fine, no worries.

The weird part that happens is when I go to use any portable radio and a repeater site that has echolink. When the node ID is put in for mine via DTMF, it will say that it's unavailable. As well, I had tried it with a portable locally and tried to dial myself--it immediately IDs and then quickly disconnects thereafter.  I am obviously missing something but I cannot spot what this setting or option may be. as this happened to anyone else?

sm3sgp, 2017-03-08:
Sound like a connection issue. If I remember correct there are problems connecting to echolink nodes from the same ip address. Maybe your phone and your svxlink are behind the same router at home?
What do you see in the svxlink logfiles?

ajkatz83, 2017-03-08:
I think you may be right. After I opened the ports for echolink, the node number changed. I didn't think that they could do that in echolink, but the node ID changed and when we connected up via RF with another repeater in town that has a node, it worked just fine, from radio to radio. I htink before we were trying to connect to ourselves, but see the verbose from svxlink saying that it couldn't connect to itself was good indication. Using the Echolink app on my phone I can still connect to all nodes in town, and it appears that is only one way. When using the DTMF with the radios to activate the node it works fine, all I changed was the ports that is required and opened them. I guess things are operating normally.

closed 2017-03-08 by sm0svx