sm0svx/svxlink #256: En-/Decoder in one class?

dl1hrc, 2017-03-12:
For some reason it may be unfavorable to create separate encoder and decoder instances, e.g. if you have a piece of hardware with en/decoder on the board and the communication is flowing over one single usb port with appropriate control messages. I try to combine both but would like to get some hints:

I would create a "recoder" class:
class AudioRecoder : public AudioSource, public AudioSink, public sigc::trackable

the class includes all methods of AudioSink+AudioSource classes. If the instance is created of course it is source and sink at the same time. I must rename/create new signals methods since e.g. allEncodedSamplesFlushed is a signal in the decoder and normal function in the encoder. Do you see problems here? It's running but before I continue with deeper changes I would like to know if you have an better suggestion.

73 Adi/DL1HRC

sm0svx, 2017-03-12:
I'm not sure what you are trying to do. Is it like a DV-dongle or something like that you are talking to?
There are some cases in SvxLink that may be of a similar problem like RX and TX sharing a serial port (AsyncSerial and AsyncSerialDevice) or multiple RX:es and TX:es sharing a single audio device (AudioIO and AudioDevice).

Maybe a AudioCodecDevice class in combination with AudioEncoder and AudioDecoder. It's hard to tell since I don't know exactly what you are trying to achieve.

dl1hrc, 2017-03-12:
> I'm not sure what you are trying to do. Is it like a DV-dongle or 
> something like that you are talking to?
>

correct. I'm not sure what happens if you open the same usb device 
twice, from decoder instance and from encoder instance. Both send 
different configuration params to it and so on...
With the serial port it seem to be no problem.

sm0svx, 2017-03-12:
That's what the XxxDevice classes are for. That class open the DV-device once and then you can instantiate one decoder and one encoder that talk to the AudioCodecDevice class. Look at the AudioDevice class for inspiration. It use a "parameterized singleton" to see if a device have been opened already and if so it returns the same instance instead of creating a new one. It's the same idea with the AudioDevice class.

So, when you create an AudioDecoderDV it will use some static function in AudioCodecDV to create a new instance of that device class and register the AudioDecoderDV instance with it. Then when you create the AudioEncoderDV instance you do the same thing but then the static function in AudioCodecDV will see that there already is a device class instance created for the given device name and return that instead.

sm0svx, 2017-03-12:
The AudioDevice class is a bit complex since it also use a factory class when creating device instances to be flexible. The SerialDevice class is a bit simpler. I hope you get the idea.

dl1hrc, 2017-03-13:
ok, thank you. I think I understand it and found the line in the class
definition too :)
Before I dive deeper, you would accept this kind of developments (if
it's working well) someday?

> The AudioDevice class is a bit complex since it also use a factory
> class when creating device instances to be flexible. The SerialDevice
> class is a bit simpler. I hope you get the idea.
>

sm0svx, 2017-03-14:
Depends on what you want to do. It's a bit fuzzy right now for me. I have had the idea myself to implement support for digital voice in SvxLink but to really make it good the whole audio framework would have to be rewritten so that the audio can be retained in coded form from RX to TX. I'm not ready to do that kind of change right now though.

If your thought is to implement like a DStarRx and DStarTx within the current framework that would probably be totally ok. That is, the RX decode to "normal" samples and the TX encode it again. I'm not sure what that will do to the audio quality though. At least it's a start!

One thing that I wonder is what are you going to use for modulation and demodulation of the digital signal?

It would be a good thing if you could publish your branch to GitHub and create a pull request quite early so that I can follow the development. Keep me posted. Maybe I need to buy some digital gear now ;-)

johnhays, 2017-03-14:
If you make your encoder/decoder so it can use AMBEserver, it is simply UDP to/from the AMBEserver which handles talking to a ThumbDV or PiDV to handle the encode/decode between AMBE and PCM.

If you want to simply repeat D-STAR, then no encoder decoder is necessary, but to cross mode to analog will require it.

Contact me direct if needed?  

https://nw-digital-radio.groups.io/g/ambe/wiki/AMBEserver-Addendum#Quick-Build-on-Linux
https://nw-digital-radio.groups.io/g/ambe/wiki/AMBE-Devices
http://nwdigitalradio.com/products/

Tobias, would you contact me off list?

dl1hrc, 2017-03-14:
yes, I have such a ThumbDV stick and will try to access it. The other
task will be to access the digital network(s), here I've started to
implement the protocols and contact developers. It's in a very early
state and will take some time to report good news :)

> If you make your encoder/decoder so it can use AMBEserver, it is
> simply UDP to/from the AMBEserver which handles talking to a ThumbDV
> or PiDV to handle the encode/decode between AMBE and PCM.
>
> If [...]

sm0svx, 2017-03-14:
It's still unclear for me. What parts do you want to implement?  Is it the RF side or the "Internet side"?

dl1hrc, 2017-03-14:
e.g.
Analog(RF) <-> Soundcard <-> AMBE-De/Encoder <-> digital Network(e.g. BM)

DMR(RF)<->Soundcard<->SvxLink+AMBE-De/Encoder<->Soundcard<->Analog(RF)

For digital NW you need a protocol, e.g. Rewind or IPSC HB, both are udb
with simple authentication

Am 14.03.2017 um 22:19 schrieb Tobias Blomberg:
>
> It's still unclear for me. What parts do you want to implement? Is it
> the RF side or the "Internet side"?
>
> —
>

dl1hrc, 2017-03-15:
BTW: dl1hrc_last_comment.replace(quote.find("udb"), 3, "UDP");

Here are some docs
Rewind protocol: https://wiki.brandmeister.network/index.php/Simple_External_Application
IPSC HB (old version from 2015), as far as I know the new specs will be published soon:
http://www.qsl.net/kb9mwr/projects/dv/dmr/DMRplus_IPSC_Protocol_for_HB_repeater.pdf

sm0svx, 2017-03-15:
OK, thanks. Then it's a bit more clear what parts you want to work on. It's something like that I had in mind as well so that's a good start :-)

How do you plan on doing demodulation and protocol decoding of the digital RF signal?

Can you tell me something more or point me to some information about "Rewind", "IPSC HB" and "udb"? New terms for me.

Am I correct in assuming that DMR use the same audio codec as D-STAR, so both could be implemented using the same dongle?  I may get my hands on some DMR-gear soon so that would be very interesting to get going.

johnhays, 2017-03-15:
On Wed, Mar 15, 2017 at 3:54 PM, Tobias Blomberg <notifications@github.com>
wrote:

>
> Am I correct in assuming that DMR use the same audio codec as D-STAR, so
> both could be implemented using the same dongle? I may get my hands on some
> DMR-gear soon so that would be very interesting to get going.
>
>
>
Tobias, the AMBE-3000 chip (as in the ThumbDV and other newer product)
supports AMBE+ for D-STAR and AMBE+2 for DMR and you can switch between the
two easily through a data packet protocol over serial.  The original D-STAR
 'Dongle' does not support DMR AMBE+2.

https://www.youtube.com/watch?v=eOYio06rsuo
-- 

------------------------------
John D. Hays
K7VE

PO Box 1223, Edmonds, WA 98020-1223
   <http://k7ve.org/blog>  <http://twitter.com/#!/john_hays>

sm0svx, 2017-03-15:
Ok, John. Thanks for the clarification.

dl1hrc, 2017-03-16:
> OK, thanks. Then it's a bit more clear what parts you want to work on. It's something like that I had in mind as well so that's a good start :-)
> 
> How do you plan on doing demodulation and protocol decoding of the digital RF signal?

9k6 radio, soundcard or external hardware. I think there are some options.
But I think it's a good idea to start with a simple gateway DMR(Network) <-> Analog(RF)

> Can you tell me something more or point me to some information about "Rewind", "IPSC HB" and "udb"? New terms for me.
> 

udb -> UDP, I mean the connectionless protocol. Rewind and IPSC are protocols based on the UDP protocol, pse use the links above for further information.
afaik IPSC has been developed by the DMR+, Rewind by Brandmeister crew. Have already made a simple communication by ipsc, for Rewind I still need a host for testing and some feedback.