sm0svx/svxlink #259: AsyncTcpClient unable to handle unreliable internet connection

dh1dm, 2017-03-22:
in line 299 the assert(sock == -1) will fire when network stack timeout is longer than re-connect timeout
this frequently happens in 3G/4G networks, submittet patch was not accepted
issue is critical since terminates svxlink because assertions are enabled even in release builds

sm0svx, 2017-03-23:
Thanks! I'll have a look at it.

sm0svx, 2017-09-21:
In what application/part of SvxLink have you seen this problem?

sm0svx, 2017-12-18:
@dh1dm, I'm not sure where you observed this behavior. Was it with NetRx/NetTx?  Is the problem still there?  The networking code has been somewhat rewritten since this bug was reported.

dh1dm, 2017-12-20:
I think it's not a problem of the AsyncTcpClient, but the EchoLinkProxy connection handling over an unreliable 3G mobile connection. This has been re-written meanwhile, but since I'm unable to have physical access to this repeater during winter time, I can't provide you with test results.

sm0svx, 2017-12-21:
Ok. So what is the end effect of the problem?  Does SvxLink stop connecting to the proxy server when this occurs, so that it ends up in a permanent disconnected state?

dh1dm, 2017-12-27:
> 
> 
>     Ok. So what is the end effect of the problem? Does SvxLink stop connecting to the proxy server when this occurs, so that it ends up in a permanent disconnected state?

Yes, I think so, but I don't have remote access to the system, since it uses a non-public mobile IP address.

The proxy connection seems permanently broken, but the APRS connection is still working.

sm0svx, 2019-07-15:
I'm closing this since there's been no activity for quite a while. Please reopen issue if you still think it's a bug in SvxLink.

closed 2019-07-15 by sm0svx