sm0svx/svxlink #262: DTMF from external source

isladelobos, 2017-03-24:
Hi.

SvxLink can decode DTMF really well from the mic keypad or from the radio memory but if the DTMF is played from a external source like a phone near the microphone with a DTMF generator software, it not decode some numbers like (1,2,3, and some times 4) it can decode fine the rest of keypad tones.

How adjust the receiver for these first tones?

Regards.

sm3sgp, 2017-04-04:
I guess the DTMF tones get distorted in some way. Maybe some filtering in your radio or microphone.
You could probably use some external tool like audacity and record your DTMF and make some analysis. Also have a look at the DTMF_MAX_FWD_TWIST and DTMF_MAX_REV_TWIST parameters described in the the manual.

sm3sgp, 2017-04-09:
When scrolling through the manual (man svxlink.conf) I found the DTMF_DEBUG parameter.
Is is described in the local receiver section. I have not tried it myself yet:

> DTMF_DEBUG

> Set to 1 to continuously print software DTMF decoder decision parameters. This should only be  used  for  a  short while to pinpoint problems with the DTMF decoding since it will print one row of analysis parameters 100 times per second. The following parameters are printed.

              ·   pwr - The power in the audio signal. Must be over about -50dB.

              ·   q - Quality. Should be close to 1.00 for a good detection.  If the signal is strong but the value is low  any‐
                  way, the signal probably is distorted for some reason. The input audio level may be too high for example.

              ·   twist  -  The  amplitude difference between the two tones.  Should be around 0dB, which means the tones should
                  ideally be of the same strength. By default, values between -6dB to +8dB are accepted but the  thresholds  can
                  be set using the DTMF_MAX_FWD_TWIST and DTMF_MAX_REV_TWIST configuration variables.

              ·   rowq - Quality of the row (low group) tone. Should be close to one.

              ·   colq - Quality of the column (high group) tone. Should be close to one.

              ·   digit - The digit mapped to the two detected tones.

              ·   row3rd  -  The  row  tone relation to its third overtone.  Should be close to zero. If it's not, the signal is
                  probably distorted.

              ·   col3rd - The column tone relation to its third overtone.  Should be close to zero. If it's not, the signal  is
                  probably distorted.

              ·   im  -  The  relation of the two tones to their intermodulation product.  Should be close to zero. If it's not,
                  the signal is probably distorted.
`

sm3sgp, 2019-09-10:
DTMF decoding works fine in svxlink. It should be possible to troubleshoot with the DTMF_DEBUG parameter described in the manual.

closed 2019-09-10 by sm3sgp