sm0svx/svxlink #264: GPIO not being exported with the new script

kb3vgw, 2017-04-04:
the change you made to my systemd the gpio are now not getting exported on a service restart.

kb3vgw, 2017-04-04:
/lib/systemd/system/svxlink_gpio_setup is not working

sm0svx, 2017-04-06:
Ok. Do you mean like after a configuration change to the GPIO pins that you want to restart and get the new configuration loaded?

kb3vgw, 2017-04-13:
found the issue was not svxlink related it was orp related. they are not ouputing the new path to the gpio.conf

kb3vgw, 2017-04-13:
issue not svxlink code related

kb3vgw, 2017-04-13:
root@orp:~# bash -x /usr/sbin/svxlink_gpio_up
+ GPIO_CONF=/etc/svxlink/gpio.conf
+ '[' -r /etc/svxlink/gpio.conf ']'
+ . /etc/svxlink/gpio.conf
++ GPIO_PATH=/sys/class/gpio
++ GPIO_IN_HIGH=
++ GPIO_IN_LOW=
++ GPIO_OUT_HIGH=
++ GPIO_OUT_LOW=
++ GPIO_USER=svxlink
++ GPIO_GROUP=daemon
++ GPIO_MODE=0664

even with setting the path in the /etc/default/svxlink
# The user to run the SvxLink server as
RUNASUSER=svxlink

# Specify which configuration file to use
CFGFILE=/etc/openrepeater/svxlink/svxlink.conf

# Environment variables to set up. Separate variables with a space.
ENV="ASYNC_AUDIO_NOTRIGGER=1"

# AUDIO_SETTINGS_ONSTART="<string>"
# AUDIO_SETTINGS_ONSTOP="<string>"
#   <string>: shell script command(s)

if [ -r /etc/openrepeater/svxlink/gpio.conf ]; then
        . /etc/openrepeater/svxlink/gpio.conf
fi

its not reading the correct path to the gpio conf file
root@orp:~# bash -x /usr/sbin/svxlink_gpio_up
+ GPIO_CONF=/etc/svxlink/gpio.conf
+ '[' -r /etc/svxlink/gpio.conf ']'
+ . /etc/svxlink/gpio.conf
++ GPIO_PATH=/sys/class/gpio
++ GPIO_IN_HIGH=
++ GPIO_IN_LOW=
++ GPIO_OUT_HIGH=
++ GPIO_OUT_LOW=
++ GPIO_USER=svxlink
++ GPIO_GROUP=daemon
++ GPIO_MODE=0664

even with setting the path in the /etc/default/svxlink
# The user to run the SvxLink server as
RUNASUSER=svxlink

# Specify which configuration file to use
CFGFILE=/etc/openrepeater/svxlink/svxlink.conf

# Environment variables to set up. Separate variables with a space.
ENV="ASYNC_AUDIO_NOTRIGGER=1"

# AUDIO_SETTINGS_ONSTART="<string>"
# AUDIO_SETTINGS_ONSTOP="<string>"
#   <string>: shell script command(s)

if [ -r /etc/openrepeater/svxlink/gpio.conf ]; then
        . /etc/openrepeater/svxlink/gpio.conf
fi

its not reading the correct path to the gpio conf file is seems not to look and read the defaults file .

kb3vgw, 2017-04-14:
ok the /usr/sbin/svxlink_gpio_up seems not to be reading the redirect from /etc/default/svxlink file. and only looks at the /etc/svxlink/gpio.conf. there for it fails to load any other file if reassigned to a diff directory

kb3vgw, 2017-04-14:
part 1 of the issue is you brroke the svxlink_gpio_up where it says $PIN it should say gpio$pin as linux looks for the word gpio+pin nmber to configure the pin . I have tested and confirmed this issue. so this breaks odrois/pi's/ and other arm boards that follow the normal gpio layout. 

2nd is that its not reading the -r redirect in the default file because your hard setting the path in the gpio.conf with GPIO_CONF= and its not allowing a override so there for it cant read from any other configured dir.

kb3vgw, 2017-04-14:
so at this point the /lib/systemd/system/svxlink_setup_gpio is not reading the /etc/default/svxlink and getting any override to the path where it should read the gpio.conf. and its only reading /etc/svxlink/gpio.conf.

kb3vgw, 2017-04-14:
in the svxlink_setup_gpio
GPIO_CONF="@SVX_SYSCONF_INSTALL_DIR@/gpio.conf"
[ -r "$GPIO_CONF" ] && . "$GPIO_CONF"

when installed sets it to /etc/svxlink/gpio.conf  and does not read the change in the /etc/default/svxlink override file.

kb3vgw, 2017-04-15:
ok after testing it seems the svxlink_gpo_setup.service is not doing its job either. it seems not to sun the scripts

kb3vgw, 2017-04-28:
edit the svxlink_gpio_up.in
svxlink_gpio_down
replace the 
first if statement with this and it fixes the gpio . 
if [ -r /etc/default/svxlink ]; then
. /etc/default/svxlink
else
. /etc/svxlink/gpio.conf
fi

The issue was the system is not passing the defaults from the systemd  /etc/default/svxlink file to the script so this makes the script look for the default file and read it .

Bamm gpio now work like they should

sm0svx, 2017-05-13:
Why do you want it to read from both those files?  Isn't that kind of confusing? What if you are running a RemoteTrx and not a SvxLink node?  Doesn't make sense to read the config from the svxlink systemd default config file. Support for both SvxLink and RemoteTrx was tha reason I split out the GPIO part into its own service.

So I guess all this boils down to the question; why isn't it enough to just use gpio.conf?

gschwiter, 2017-06-02:
can someone tell me how to correct the issue, i have been searching all over the net this is the only thing the will patch the system every time it restarts. 
 echo "18" >/sys/class/gpio/export && echo "out" >/sys/class/gpio/gpio18/direction
the paths ../../devices/platform/soc/3f200000.gpio/gpio/gpio18
/sys/devices/platform/soc/3f200000.gpio/gpio
are locked and can not be updated with change mod.

kb3vgw, 2017-06-02:
I have to get the fix in to the src but if you use the new /etc/svxlink/gpio.conf and the systemdit will setup the gpio.

I have fixed the issue on my arm pkgs to work with svxlink and the openrepeater gui. I need to repush the fix up later today.

I have been busy moving into my new place. And it took away from svxlink dev time

If not using system there is the svxlink_gpio_up and down scripts that read from the gpio.conf file and configure the gpio.

From: gschwiter
Sent: Friday, June 2, 2017 7:37 AM
To: sm0svx/svxlink
Cc: Richard Neese; State change
Subject: Re: [sm0svx/svxlink] GPIO not being exported with the new script(#264)

can someone tell me how to correct the issue, i have been searching all over the net this is the only thing the will patch the system every time it restarts.
echo "18" >/sys/class/gpio/export && echo "out" >/sys/class/gpio/gpio18/direction
the paths ../../devices/platform/soc/3f200000.gpio/gpio/gpio18
/sys/devices/platform/soc/3f200000.gpio/gpio
are locked and can not be updated with change mod.

gschwiter, 2017-06-03:
Aloha 
I'm not sure if I can reply back through the email, however thanks for the fast reply. I think I will wait for the fix. I created a work around. I have been learning a lot over the past week and really like the software. I have it installed on a small  experimental repeater in Hawaii on the Big Island. It's amazing what can be done with the software. 
Thanks again,
Gary Schwiter
WH6EPS

kb3vgw, 2017-06-03:
What hardware are you using ?  is it intel or arm based ?

From: gschwiter
Sent: Friday, June 2, 2017 10:04 PM
To: sm0svx/svxlink
Cc: Richard Neese; State change
Subject: Re: [sm0svx/svxlink] GPIO not being exported with the new script(#264)

Aloha 
I'm not sure if I can reply back through the email, however thanks for the fast reply. I think I will wait for the fix. I created a work around. I have been learning a lot over the past week and really like the software. I have it installed on a small experimental repeater in Hawaii on the Big Island. It's amazing what can be done with the software. 
Thanks again,
Gary Schwiter
WH6EPS

gschwiter, 2017-06-03:
Aloha 
I am currently running this on a RPI3. I was thinking of running/test it on some old WISE terminals however that will be down the road. I hope your done packing and moving. 
Thanks again for helping with this issue. On a side note do you manage any repeaters if so what modules do you find useful. 

My overall goal is to combine this with pi weather station and asterisk outbound sip calls "outbound phone patch" I was also looking at the SvxLink.de page and noticed a lot more features. 2 that caught my eye were Storm warning and eMail-2-RF. I'm still having issues with the Metar, do you know if they are using TXT or XML now I understand there was a change. I'm getting no such airport. I believe it is a server connection issue.

Thanks Again for all the help.
Gary Schwiter
WH6EPS

kb3vgw, 2017-06-09:
I have started working on new pkgs I have new armhf and arm64 pkgs now in my devel branch that have the new reflector module. I will have amd64 and i386 in later today/tomorrow. That will make installs cleaner and easier.

I will also update my script to support this . I will be working on the menu starting next week to try and make it mor updated with all the new functions.

If you look at the new svxlink.conf in my pkg there is a lot more for use in it now.

Rich
Kb3vgw

From: Gary Schwiter
Sent: Saturday, June 3, 2017 5:58 PM
To: sm0svx/svxlink
Cc: Richard Neese; State change
Subject: Re: [sm0svx/svxlink] GPIO not being exported with the new script(#264)

Aloha 
I am currently running this on a RPI3. I was thinking of running/test it on some old WISE terminals however that will be down the road. I hope your done packing and moving. 
Thanks again for helping with this issue. On a side note do you manage any repeaters if so what modules do you find useful. 

My overall goal is to combine this with pi weather station and asterisk outbound sip calls "outbound phone patch" I was also looking at the SvxLink.de page and noticed a lot more features. 2 that caught my eye were Storm warning and eMail-2-RF. I'm still having issues with the Metar, do you know if they are using TXT or XML now I understand there was a change. I'm getting no such airport. I believe it is a server connection issue.

Thanks Again for all the help.
Gary Schwiter
WH6EPS

kb3vgw, 2017-06-09:
I have new arm pkgs for armhf and arm64 in my repo for testing this has the new reflector function in it.

I am working n amd64 and i386 pkhs for making reflectors on machines

From: Gary Schwiter
Sent: Saturday, June 3, 2017 5:58 PM
To: sm0svx/svxlink
Cc: Richard Neese; State change
Subject: Re: [sm0svx/svxlink] GPIO not being exported with the new script(#264)

Aloha 
I am currently running this on a RPI3. I was thinking of running/test it on some old WISE terminals however that will be down the road. I hope your done packing and moving. 
Thanks again for helping with this issue. On a side note do you manage any repeaters if so what modules do you find useful. 

My overall goal is to combine this with pi weather station and asterisk outbound sip calls "outbound phone patch" I was also looking at the SvxLink.de page and noticed a lot more features. 2 that caught my eye were Storm warning and eMail-2-RF. I'm still having issues with the Metar, do you know if they are using TXT or XML now I understand there was a change. I'm getting no such airport. I believe it is a server connection issue.

Thanks Again for all the help.
Gary Schwiter
WH6EPS

gschwiter, 2017-06-11:
Aloha 
I was still having some issues getting the script to run. It was some dependency issues. I Think I'm going to start over from the beginning, for the pi 3 what is the recommend best setup and setup script. I think I'm going to scrap the OpenRepeater and just use the SVXLink. 
Thanks
Gary

kb3vgw, 2017-06-11:
I will be posting a img today for just svxlink. Give me 2 hours to get things done. I will then explain to you how it works with system . it will also have the new svxreflector function in it for testing

From: Gary Schwiter
Sent: Sunday, June 11, 2017 6:34 AM
To: sm0svx/svxlink
Cc: Richard Neese; State change
Subject: Re: [sm0svx/svxlink] GPIO not being exported with the new script(#264)

Aloha 
I was still having some issues getting the script to run. It was some dependency issues. I Think I'm going to start over from the beginning, for the pi 3 what is the recommend best setup and setup script. I think I'm going to scrap the OpenRepeater and just use the SVXLink. 
Thanks
Gary

kb3vgw, 2017-06-11:
https://github.com/kb3vgw/svxlink-install-scripts/blob/master/debian-unified-svxlink-install.sh

Use the devel branch for scvxlink. Then to setup gpio add the gpio to /etc/svxlink/gpio.conf and svxlink.conf  .

They use service svxlink restart. This should then setup and configure the gpio pins . and you should have a working system.

I will be working on the shell menu this week to bring it upto par.

From: Gary Schwiter
Sent: Sunday, June 11, 2017 6:34 AM
To: sm0svx/svxlink
Cc: Richard Neese; State change
Subject: Re: [sm0svx/svxlink] GPIO not being exported with the new script(#264)

Aloha 
I was still having some issues getting the script to run. It was some dependency issues. I Think I'm going to start over from the beginning, for the pi 3 what is the recommend best setup and setup script. I think I'm going to scrap the OpenRepeater and just use the SVXLink. 
Thanks
Gary

kb3vgw, 2017-06-11:
The repo has armhf / arm64/intel i586 / amd64 currently for testing

From: Gary Schwiter
Sent: Sunday, June 11, 2017 6:34 AM
To: sm0svx/svxlink
Cc: Richard Neese; State change
Subject: Re: [sm0svx/svxlink] GPIO not being exported with the new script(#264)

Aloha 
I was still having some issues getting the script to run. It was some dependency issues. I Think I'm going to start over from the beginning, for the pi 3 what is the recommend best setup and setup script. I think I'm going to scrap the OpenRepeater and just use the SVXLink. 
Thanks
Gary

kb3vgw, 2017-06-11:
Just fixed issues in script and made a img I will post it in a min in my github repo

From: Gary Schwiter
Sent: Sunday, June 11, 2017 6:34 AM
To: sm0svx/svxlink
Cc: Richard Neese; State change
Subject: Re: [sm0svx/svxlink] GPIO not being exported with the new script(#264)

Aloha 
I was still having some issues getting the script to run. It was some dependency issues. I Think I'm going to start over from the beginning, for the pi 3 what is the recommend best setup and setup script. I think I'm going to scrap the OpenRepeater and just use the SVXLink. 
Thanks
Gary

kb3vgw, 2017-06-11:
https://github.com/kb3vgw/Svxlink-Images/releases/download/1.5.99-8/svxlink-raspbian-devel-6-11-17.zip 

I am working on pkg updates I fixed the service files. The reflector service file was wrong . once I push updates I will let you knw.

From: Gary Schwiter
Sent: Sunday, June 11, 2017 6:34 AM
To: sm0svx/svxlink
Cc: Richard Neese; State change
Subject: Re: [sm0svx/svxlink] GPIO not being exported with the new script(#264)

Aloha 
I was still having some issues getting the script to run. It was some dependency issues. I Think I'm going to start over from the beginning, for the pi 3 what is the recommend best setup and setup script. I think I'm going to scrap the OpenRepeater and just use the SVXLink. 
Thanks
Gary

RichNeese, 2017-08-17:
Ok the new functions are there except for new node password update and the the script update option .

Both will be done tomorrow an then that’s the basic menu done we can then slowly grow off it and be warned this can make the menu big as we add things

Sent from Mail for Windows 10

RichNeese, 2017-11-22:
I have update the scripts and fixed what some one else broke. I have tested on the pi/nanopi/odroid and it is working properly I have pushed it in my latest change list to tobias

sm3sgp, 2019-09-10:
GPIO configuration should work fine in latest release (and previous release as well).

closed 2019-09-10 by sm3sgp