sm0svx/svxlink #267: SVXReflector ACCEPT/REJECT?

sq4bja, 2017-04-17:
Don`t you think that a useful SVXReflector function could be ACCEPT/REJECT in SVXReflector.conf? We have failed repeater connected to SVXReflector. We do not have access to it. To lock it we have to change the password in svxreflector.conf and in a dozen other repeaters. It's very laborious.
Another example. The reflector password has leaked somewhere, and someone has illegally connected his repeater. How to lock it?
And here is a solution from Echolink. White list and black list. ACCEPT CALLS, REJECT CALLS.
It would be very useful. To think in your free time.

Tom SQ4BJA

f5nlg, 2017-05-07:
i use Drop from iptables 
iptables -A INPUT -s IPADDRESS -j DROP
and that solve the problem for unwanted repeter ...

sq4bja, 2017-05-08:
Ok, but only, when every repeater have static IP. My
          repeaters do not have this.
    
    W dniu 2017-05-08 o 00:37, f5nlg pisze:
    
    
      i use Drop from iptables
        iptables -A INPUT -s IPADDRESS -j DROP
        and that solve the problem for unwanted repeter ...
      —
          it on GitHub, or mute
          the thread.

f5nlg, 2017-05-08:
So you put dynamic dns on all identified repeater and you accept only 
this with :

iptables -A INPUT -s repeater1.dynamic.dns -j ACCEPT

the others will be not accepted ...

don't forget your IP adress it's a verry basic usage of iptables ...

sm0svx, 2017-05-13:
@f5nlg: That will only work as long as the node keeps it's IP since no matter if you give iptables a hostname or IP address it's the IP address that will be stored. Depending on the situation it may be enough though.

My intension is to implement a per user (per callsign) password. Then it will be possible to exclude just one node, whatever the reason is. That will be much safer than blocking the IP or a specific callsign.

sm3sgp, 2019-09-10:
Password per user has been available since 17.12 release.

closed 2019-09-10 by sm3sgp