sm0svx/svxlink #268: Metar Info

Jonthetrombonist, 2017-04-29:
Hello all, I'm new to linux, and can just get around files paths and modify files with Nano. I'd like to update to the trunk to allow me to use the updated Metar Info, since NOAA stopped their services. Could someone list the steps out for me?  73 K5DVT

Jonthetrombonist, 2017-05-05:
(Sorry for the long reply) I'm running Debian on a raspberry pi b+

Jonthetrombonist, 2017-05-06:
I'd be interested in the install script

dl1hrc, 2017-05-07:
Ok. Sri. I'm not at home at the moment. Will have better internet access this evening. 73s Adi

dl1hrc, 2017-05-07:
Hi,
I've changed the script a bit so that you will get English messages. If the installation has a bit more time, I can change it for you to have a complete en_US environment.
It was primary written for Germans, will be back next week and then better hardware and internet access :)
The ModuleMetarInfo.conf has been extended, do not forget to add the params TYPE, SERVER and LINK to your config if already installed on your system
Here is a working ModuleMetarInfo.conf:

[ModuleMetarInfo]
NAME=MetarInfo
ID=5
TIMEOUT=120
TYPE=TXT
SERVER=tgftp.nws.noaa.gov
LINK=data/observations/metar/stations
LONGMESSAGES=1
REMARKS=1
DEBUG=1
AIRPORTS=EDDP,EDDM,EDDF,EDDS,SKSM,SKBO,KJAC,KTOL

Here is the way to get and start the script, do the following on your destination system as root:

wget http://svxlink.de/download/svx-install_en.sh
chmod 750 svx-install_en.sh
./svx-install_en.sh

The script install the newest trunk on your system including all the needed development files. It may take some time depending on your hardware.
The script creates an logfile, in the case of errors/problems you can send me the installation.log and we can solve the problem step-by-step, ok?

73s de Adi

sm3sgp, 2019-09-10:
Hopefully the configuration of Metar Info was sucessful. Use the groups.io mailing list for further discussions on this topic.

closed 2019-09-10 by sm3sgp