sm0svx/svxlink #271: Public & Private IP Adresses

David6FEO, 2017-05-02:
I am the keeper of GB3MW which is now running SVXlink, and the users seem very pleased.
We have noticed a problem with Public and Private IP addresses, MW has a public IP address and also a 192.168 address which is part of my network, GB3HT also has a different public IP address and a 192.168 address which is also part of my network.
When EchoLink tells one repeater to connect to the other via their public IP's they fail as they try to connect via the shortest route their 192.168 addresses.
Has anybody had this problem previously and give me some pointers as to how to solve this problem.
I am confident this is not an SVX problem.
73, David.
G6FEO.
Keeper GB3WK & GB3MW.
p.s GB3WK is also going to become an SVXlink repeater.

sm0svx, 2017-05-13:
The BIND_ADDR configuration variable probably will solve your problem. From the manual page:

> **BIND_ADDR**
> This  configuration  variable  can  be set to bind the EchoLink network sockets to a
> specific IP-address/interface. This may be needed if the  computer  is  fitted  with
> more than one ethernet interface and EchoLink should only be used on one of them.
> 
> Example: BIND_ADDR=192.168.0.1

Set it to your public IP address to solve the problem.

pe1chl, 2017-07-18:
You are correct that it is not an SVXlink problem.  It is very important that an Echolink repeater has only a single address and that has to be the address as seen by the Echolink directory server.
So, when you have NAT between your repeater and echolink.org you HAVE to use the NAT'ted address as seen by echolink.org for all communication, even internally (e.g. via a hairpin NAT in your router).   When you start qtel (the SVXlink echolink client) you can see that address in the directory listing.
While it is possible to use BIND_ADDR to force a source address for the outgoing connect, e.g. when your system is multi-homed, it is not a solution that allows you to have alternative addresses for internal and external access.  That simply cannot be done with Echolink.

sm3sgp, 2019-09-10:
No issue with svxlink.

closed 2019-09-10 by sm3sgp