sm0svx/svxlink #272: SimplexLogic Roger beep doesn't work on 1.5.99.7 ( SVXREFLECTOR )

f5nlg, 2017-05-07:
Any solution to have a working roger beep on SimplexLogic with SvxReflector , it seems to have a bug somewhere but i can't found where ... 
it was working on previous versions ( with echolink module on logic with fonctionnal roger beep )

and how to have roger beep on multiple logics ? 3 simplexLogic for example ?

thanks in advance

f5nlg, 2017-05-10:
Nobody is interested by my subject ??? have i miss something ?

sm0svx, 2017-05-13:
Works here on SvxLink v1.5.99.9-svxreflector. The EchoLink logic need to be active on the logic core you are transmitting into, of course, for the roger beep to be played back. And it is only played back on the node that received the transmission.

One other reason may be if you have done any changes to the TCL event handling scripts you may have broken something. In that case, try to go back to the default scripts and see if that solves the problem.

sp2ong, 2017-05-13:
But it will be useful has roger beep without activate echolink moule. I have cross band link 144/438 svxlink and 144 mhz is echolink node but 438 Mhz band is only svxlink simplex logic system and we don't use echolink module. On svxlink 438 Mhz roger beep don't work but in Logic.tcl look like a=is defined but I don't know why work only when we use echolink module

sm0svx, 2017-05-13:
It's simply the way it's implemented at the moment. You should see two logics as two separate systems that happen to interchange audio. It's like you had set up two SvxLink nodes on two different computers and connected an audio cable between them. One node cannot know what happens on the other node. A roger beep will only be transmitted on the node that has the EchoLink module loaded and activated.

I'm not saying that it have to be like that for ever, just that it's implemented like that right now. It's logical in some use cases and in your case it may not be logical. So a configuration option would have to be added and the roger beep transmission logic code would have to be extended.

f5nlg, 2017-05-18:
hello , an other solution was to generate the roger beep on SVXREFLECTOR so it will be transmit to all the node/repater connected to it. if it's simple to do ...

sm3sgp, 2019-09-10:
This is now available in the latest 19.09 release.

Now possible to set the roger sound to be always on in the SimplexLogic, even when no module is active. Use the RGR_SOUND_ALWAYS configuration variable to enable the feature.

closed 2019-09-10 by sm3sgp