sm0svx/svxlink #273: Long PTT delays when using the Fe-Pi and UDRC

n7ipb, 2017-05-11:
I posted about this on the mailing list back in September of last year but never got a response and never found the time to troubleshoot it myself.  I've moved on and designed my own card for the RPi with USB sound chip and gpio PTT/CSQ but since both cards are clearly being used with s

vxlink it would be nice to track this down.  

In normal repeater or echolink configurations both seem to work fine, its when svxlink is configured to have one Repeater and one Simplex logic with local rx/tx on the same interface that the issue shows up and then only when not using USB for the audio.

I've attached a copy of my original posting including the config file.

sm0svx, 2017-05-14:
Seems like a tricky problem. I can see no obvious configuration error. I have a UDRC card here but I have not tested it yet. I'll need to find the time to test your configuration.

n7ipb, 2017-05-14:
On Sunday, May 14, 2017 10:09:25 AM PDT Tobias Blomberg wrote:
> Seems like a tricky problem. I can see no obvious configuration error. I
> have a UDRC card here but I have not tested it yet. I'll need to find the
> time to test your configuration.

Thanks Tobias.

I went round and round trying to find the issue but finally ran out of time to 
troubleshoot.  I need to have a working system to install soon and the easiest 
way was to use a USB device since that worked.

Both the UDRC and Fe-Pi work really well for Echolink or a single Simplex or 
Repeater logic but not the combo I need because I have a link to another 
repeater (non-svxlink controlled) that needs the local simplex connection.

I still have the ability to test so feel free to ask me to test or whatever
-- 
Ken - N7IPB
Email: n7ipb@wetnet.net
JID: n7ipb@jabber.wetnet.net
PGP Sig: F42B EF90 3CD3 31C7 3056  122E 993A 7B2E 5138 C42A 
“I never am really satisfied that I understand anything; because, understand 
it well as I may, my comprehension can only be an infinitesimal fraction of 
all I want to understand” -Ada Lovelace

kb3vgw, 2017-05-16:
to use the fe-pi ypu have to set the sounds to 48000. this fixes the audio issues

Dloranger, 2017-06-05:
I ran into N7IPB at Seapac convention and we discussed what he was seeing and we are still observing this on the ICS platform as well.  I filed the bug under the ORP tracker, but this appears to be the exact same issue.  

We witnessed delays ranging from ~100ms to almost a second, the delay appears to be random.

sm0svx, 2017-12-18:
Is this still a suspected problem with SvxLink or is it a hardware problem? I'm a bit confused about if it's a problem with UDRC or Fe-Pi or when combining them.

What's the "ICS platform"?  "ORP tracker"?

Dloranger, 2017-12-18:
I am no longer seeing this problem, I am still not sure why the delay was so random, but I suspect it has to do with the 1-wire GPIO  on the raspberry pi which is used as a temp sensor.  It seems to choke the system for about 1-3 seconds (proc = 100%).

The "ICS platform" is a commercial board, that was previously unnamed. link below for reference
http://www.ics-ctrl.com/PI-REPEATER

n7ipb, 2017-12-19:
On Monday, December 18, 2017 12:53:49 PM PST Tobias Blomberg wrote:
> Is this still a suspected problem with SvxLink or is it a hardware problem?
> I'm a bit confused about if it's a problem with UDRC or Fe-Pi or when
> combining them.

I don't believe it was a hardware problem since both the UDRC and Fe-Pi work fine in other 
applications.  I tested both of them with a GPIO for PTT/SQ and saw the delays.  

I ended up designing my own Pi hat USB sound chip interface for my repeaters  and haven't 
seen the issue.

I've been planning on testing with the UDRC again but haven't had the time.  For some reason 
when I put the first of three Svxlink systems on the air back in July there was an explosion of 
local interest and I've been busy putting together a plan for others to duplicate. It looks like 
we'll have at least three more repeaters next year to link to the current three.

I did talk to a Kenny (KU7M) at the local breakfast gathering last Saturday and he's getting set 
up to test the UDRC with Svxlink on his 1.2Ghz repeater.  That should help us identify if the 
problem is still there  but I don't expect an answer for some time yet.

If Dloranger isn't seeing it and you can't duplicate it I'm happy with you closing the issue.  It's 
not like it can't be reopened if additional information becomes available.

-- 
Ken - N7IPB
Email: n7ipb@wetnet.net
JID: n7ipb@jabber.wetnet.net
PGP Sig: F42B EF90 3CD3 31C7 3056  122E 993A 7B2E 5138 C42A 
I've calculated my velocity with such exquisite precision that I have no idea where I am.

sm0svx, 2017-12-21:
Ok. Thanks for the additional information. I'll keep the issue open for now so that it's not forgotten.

Dloranger, 2018-01-16:
Tobias,

I got an email from a customer last night that he is seeing frequent dropouts on his setup that sounds familiar to this.  I am getting my station back on the air to help work through the details of what he is seeing.

Symptom #1
Operator 1 (Transmitter) keys up and begins transmitting, first 1-2 seconds are lost to operator #2 (receiving station) (need to confirm if SQL is active or not on operator #2 radio, I believe it is)
While operator #1 transmits, operator #2 hears intermittent dropouts that last ~1-3 seconds.

Symptom #2 
During ID, the call sign playback is stopped for 1-2 seconds mid-stream

Attached is the basic configuration file
[svxlink.conf.txt](https://github.com/sm0svx/svxlink/files/1636935/svxlink.conf.txt)

If you send me your address I can send you one of the ICS hardware boards (fits on a raspberry pi 3) for a test bed.

sm0svx, 2018-01-20:
@Dloranger, thanks for the feed-back. My address is up to date on qrz.com. I don't own a pi3 at this moment but I have a pi2. Does the board work on that as well?

Dloranger, 2018-01-20:
Shipped the controller today

sm0svx, 2019-07-24:
Using the ICS controller board, for some reason the calls to the two functions snd_pcm_hw_params and snd_pcm_prepare in the Alsa API take approximately 420ms each. That is why it takes so long to open the audio device.

Changing the I2C bus clock frequency to 400kHz as suggested by @Dloranger does not seem to change anything so the communication speed to the audio chip does not seem to be the reason for the delay.

The same type of behaviour can be demonstrated using aplay. I first created an empty wav-file using sox.
```
sox -traw -esigned-integer -b16 -r48000 -c2 /dev/null /tmp/null.wav
```
Comparing the built in audio device with the ICS board give the following difference.
```
# ICS board
$ time aplay -Dhw:CARD=Audio /tmp/null.wav 
Playing WAVE '/tmp/null.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo

real    0m0.441s
user    0m0.011s
sys     0m0.010s
```
```
# Built in audio device
$ time aplay -Dhw:CARD=ALSA /tmp/null.wav 
Playing WAVE '/tmp/null.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Stereo

real    0m0.024s
user    0m0.023s
sys     0m0.000s
```
I'd say there's something fishy with the Fe-Pi/RPi combination.

sm0svx, 2019-07-24:
I've implemented a workaround for this problem. It consists of two parts: not closing the audio device once opened and feeding the playback device with zeros when there's no audio to write. Will test it and commit tomorrow.

Dloranger, 2019-07-25:
It is possible this workaround may also fix #96 and #352 depending on how you implement it.

Also with your method of demonstrating the timing delay independent of the svxlink software, I will post to the rpi foundation a bug report, perhaps they can fix the bug at some point.  Something that severe should be easy to chase down by the original driver authors I would hope.

sm0svx, 2019-07-25:
The workaround is in master now. Set AUDIO_DEV_KEEP_OPEN=1 in all local receiver RX and TX sections. Also set the environment variable ASYNC_AUDIO_ALSA_ZEROFILL=1 before starting SvxLink.

closed 2019-07-25 by sm0svx

Dloranger, 2019-10-05:
@sm0svx 

Sorry to seem dense on this one, but how do we set the ASYNC_AUDIO_ALSA_ZEROFILL=1 environmental variable?

I am guessing we want to edit the svxlink.service file to set this variable, but this is not really my expertise area.

Can we get a solution here and/or specific guidance in the svxlink manual for how to utilize this feature?

sm3sgp, 2019-10-05:
This was recently discussed on the groups.io list.
Here: https://groups.io/g/svxlink/message/221

If you use the systemd scripts you can edit the /etc/default/svxlink file.
The parameter is documented in the manual, "man svxlink"

Note, in latest git master the ASYNC_AUDIO_ALSA_ZEROFILL is set to 1 by default.

From the mailinglist:
Environment variables are inherited from the user who launch the application.

So, either:
Set the environment variable in /root/.bashrc
Set it in your start script like "export ASYNC_AUDIO_ALSA_ZEROFILL=1" (I'd say this is the preferred one)
Set it on the command line like
$ ASYNC_AUDIO_ALSA_ZEROFILL=1 svxlink --runasuser=svxlink