sm0svx/svxlink #274: SVXReflector send via events List of nodes

sp2ong, 2017-05-13:
It will be nice to have via event sources access to svxreflector list of joined nodes similar like in echolink module list of echolink nodes in procedure client_list_changed . I thing that we are need create ReleectorLogic.tcl in which will be "publishStatEvent Reflector:nodes ..."

73 Waldek

sm0svx, 2017-05-13:
The Reflector probably will never have the event handling system but some other system to get information out. The Reflector is at a very early stage at the moment and as a fact it's not even released yet but just experimental. There are still some basic functionality to work on before adding more features.

dl1hrc, 2017-06-12:
Would be nice to have the event handler in the ReflectorLogic though. Conceivable applications would be logbook's or to show the connected nodes on LCD-Display or to process the connections/messages on web pages. I fell it's better to have defined tcl procedures instead of grepping logfiles.

sm0svx, 2017-06-19:
What would be better is some web API that can be used to get all sorts of information out from SvxLink. Using the event handler for this is a hack.

sp2ong, 2017-06-20:
web API will be a nice tool to get all information but I think that Adi and Rob have more experiences with this task. At current, I have use script eventsources by Rob which allows read only one STATE PTY in the loop to display information. Maybe web API will allow collects in one time different information

sm0svx, 2017-06-20:
A good web API would make all PTYs obsolete and is a more natural way for a web application to interact with SvxLink.

closed 2022-03-22 by sp2ong