sm0svx/svxlink #275: SVXLINK Error while loading libraries

Jonthetrombonist, 2017-05-14:
In compling SVXLINK from the master trunk, then trying to run it (Both under user Pi and Sudo), I get this error:

_svxlink: error while loading shared libraries: libasynccpp.so.1.4: cannot open shared object file: No such file or directory_

I'm running this on a raspberry pi under Debian. 
Is it possible that I am missing a library? Thanks and 73 K5DVT

sm0svx, 2017-05-15:
libasynccpp is included in SvxLink so there is nothing missing that you can install. Did you follow the [installation instructions](https://github.com/sm0svx/svxlink/wiki/InstallationInstructions) exactly?

RichNeese, 2017-11-22:
This has happen on Raspbian and armbian you have to apt-get install libasynccpp1.4 libecholib1.3.
Currently the libs are ni getting added inlike they should even on debian pkging. I am working to try and track down what happen but it only happen after svxreflector was mergerd into the latest branch.
so I am reading the chnage logs and back treacking this as best I can. there is also another error
.usr/share/svxlink/events.tcl in logic SimplexLogic: missing close-brace but we can not find where the issue is.

sm0svx, 2017-11-27:
You should not install libasync and libecholib from packages (using apt) when compiling and installing SvxLink from source. That is asking for trouble.

sm3sgp, 2019-09-10:
I think this issue is resolved by now? Serveral instructions exists on the wiki for raspbian.

closed 2019-09-10 by sm3sgp