sm0svx/svxlink #279: Voter and open squelch with minus value

sp2ong, 2017-05-27:
I have use squelch detector SIGLEV and I have set
SIGLEV_OPEN_THRESH=20
SIGLEV_CLOSE_THRESH=10

I have seen in the log of svxlink that time to time Voter open squelch
with minus value:

Sat May 27 06:16:22 2017: Voter: The squelch is OPEN (Rx1=-5.30377)
Sat May 27 06:16:22 2017: Voter: The squelch is CLOSED (Rx1=-1.0125)
Sat May 27 06:16:22 2017: Tx1: Turning the transmitter ON
Sat May 27 06:16:23 2017: Tx1: Turning the transmitter OFF

Why is squelch open if I have set up that SIGLEV_OPEN_THRESH=20 ???? 

Waldek SP2ONG

sm0svx, 2017-06-19:
The most probable reason for negative values like that is that the squelch has opened briefly and then the signal has disappeared almost immediately. The value being displayed is not necessarily the same value that caused the squelch to open so what you see is a measurement from after the signal has disappeared.

SvxLink behaviour heavily depends on config and since it's very flexible it's hard to tell exactly what's causing the negative values in your case.

sm3sgp, 2019-09-10:
@sp2ong Waldek, do you still have problem with this, or can we close this issue?

sp2ong, 2019-09-11:
Too much time has passed to return now and look for this problem

closed 2019-09-11 by sp2ong