sm0svx/svxlink #281: PlayTone Enhancement, pops in CW ID

Dloranger, 2017-05-31:
I am noticing clicks in the audio between CW ID characters and I think I have isolated the issue down to failing to complete the a sine wave which results in a square edge being generated when the next character is played.

[image]

My assumptions
1) the DAC holds the last written value and does not return to zero between characters (pauses between dits/dahs)

What I believe should be done is the tone should always terminate at a zero crossing (or very close with a final forced value of zero), this will allow the next character to start very close to the last value written to the DAC to eliminate the large delta in the signal

[tone generator.xlsx](https://github.com/sm0svx/svxlink/files/1043225/tone.generator.xlsx)

DG1BBQ, 2017-06-18:
Good idea. I've been noticing this too on my hotspot (orangePi PC with SDR-dongle, and active speaker for receive audio).  It's very prominent there, because of the lack of a splatter filter that is normaly applied in the Transmitter.
On our local repeater it's normaly not as noticeable, since it has a splatter filter of course. Since we've  added a fast forward looking limiter to the audio output  though alsa, it causes  the keying of the CW  and beeps to sound a bit hard keyed.

sm0svx, 2017-06-19:
Yes. The tone playback algorithm could probably need some improvement. Thanks for reporting.