sm0svx/svxlink #287: make audio error

sonnyp, 2017-06-08:
Hello,

I'm getting the following error on make with 15.11

```
make[2]: *** [async/audio/CMakeFiles/asyncaudio.dir/build.make:831: async/audio/CMakeFiles/asyncaudio.dir/AsyncAudioDeviceAlsa.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:223: async/audio/CMakeFiles/asyncaudio.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

clem16, 2017-06-09:
I seem to be getting the exact same build error on Arch as well.
I posted about it here: https://github.com/sm0svx/svxlink/issues/284 Still no response. 
So it's not just me having this issue...

> make[2]: *** [async/audio/CMakeFiles/asyncaudio.dir/build.make:831: async/audio/CMakeFiles/asyncaudio.dir/AsyncAudioDeviceAlsa.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:226: async/audio/CMakeFiles/asyncaudio.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

sm0svx, 2017-06-19:
Solved in git. Unfortunately the released code will not compile with newer versions of the compiler.

closed 2017-06-19 by sm0svx