sm0svx/svxlink #290: svxlink documentation

alabamatoy, 2017-06-18:
Essentially, there is none.  the wiki document only addresses remote DTMF operation, and has nothing about configuration of the sysop server or how to connect it to the echolink hub or what the CLI options are.

????

sm0svx, 2017-06-19:
Admittedly the documentation could be better but it's not non-existent as you suggest.

You have the [installation instructions](https://github.com/sm0svx/svxlink/wiki/InstallationInstructions) in the wiki and links to all manual pages. The [svxlink manual page](http://www.svxlink.org/doc/man/man1/svxlink.1.html) describe all command line arguments. The [svxlink.conf manual page](http://www.svxlink.org/doc/man/man5/svxlink.conf.5.html) describe the configuration file in-depth. The [ModuleEchoLink manual page](http://www.svxlink.org/doc/man/man5/ModuleEchoLink.conf.5.html) describe the configuration of the EchoLink module.

All those manual pages should also be available at the command line on a properly installed system, e.g. "man svxlink", "man svxlink.conf", "man ModuleEchoLink" etc.

alabamatoy, 2017-06-19:
Wow, thanks!  Good info.  Looks like this will be a research project.

Do you know of anyone who has implemented this with the USB Rigblaster?  I
would very much like to learn from their mistakes.

Thanks again.

Herndon Elliott
Madison, Al
https://keyserver.pgp.com key ID: 24B60B6150130832
ΜΟΛΩΝ ΛΑΒΕ  "molon labe"

sm0svx, 2017-06-20:
No, I'm not aware if anyone is using the USB Rigblaster. The best way to get in touch with the SvxLink community is to subscribe to the svxlink-devel mailing list and ask questions there.

dl1hrc, 2017-06-20:
As far as I can see, it's a very simple adapter without hardware squelch. It's not really clear for me why they connect the soundcard input with the soundcard output directly????? But just to be sure, we're talking about this: http://www.westmountainradio.com/pdf/nomicman4.pdf
If so, you will never run a repeater with that adapter. Try the following configuration for a simplex link without guarantee:

[Tx-Section]
PTT_TYPE=SerialPin
PTT_PORT=/dev/ttySx    # x is your serial port number where you have connected your adapter, for usb use /dev/ttyUSBx
PTT_PIN=RTS

[Rx-Section]
SQL_TYPE=VOX

I really recommend to take a hole raster board, some R's and C's for less bucks and do it better.

alabamatoy, 2017-06-20:
The Nomic makes a perfect interface for echolink, VoIP into the radio.  It
works very well.  Im not trying to run a repeater, only a simplex VoIP
interface to a radio.

This is very helpful.  I will attempt to set it up, and post back how it
goes.  Thanks!

Herndon Elliott
Madison, Al
https://keyserver.pgp.com key ID: 24B60B6150130832
ΜΟΛΩΝ ΛΑΒΕ  "molon labe"

closed 2017-10-21 by sm0svx