sm0svx/svxlink #291: Question relating to sigc events

dl1hrc, 2017-06-27:
Would like to know if I can configure the event execution of a sigc routine more exactly.
E.g. Async::Serial class, Event serial->charactersReceived.connect(...);
The comment says that the event occurs if one or more characters has been received. In my application the event is triggered after 20 to 50 characters on a serial line with 460kBaud even though I send 200 bytes in one complete sequence over the serial adapter without delay.
Is there a chance to set a delay or timeout to "catch" the 20ms gap between two character streams of e.g. 200bytes?
73s de Adi / DL1HRC

sm0svx, 2017-07-18:
Basing a serial protocol on real-time constraints for a normal Linux computer is a bad idea. What you normally do is to build a frame structure so that the application can detect where a message start and end. If the message is text based a simple line-feed at the end of the command is the most common way.

It is the OS that decides how often the charactersReceived callback get called. If the serial circuit would support "timing based framing" it might work but I have never seen such a feature in PC hardware.

dl1hrc, 2017-07-18:
Ok, thanks. I've solved it with a ring buffer, detecting the length of the frame by included protocol specification (byte[2]*256 +  byte[3])
73s Adi

closed 2017-07-18 by dl1hrc

sm0svx, 2017-07-18:
Ok, good. That is a better solution although it sounds like it could get out of sync if characters are lost. If you have something like a "start of frame" indicator to sync to, or some other predictive information in each frame,  that would eliminate the problem.