sm0svx/svxlink #294: Unable to disconnect whith permanently incomming interference from connected client

DG1BBQ, 2017-07-02:
We are running a linked repeater system in the northwest of Germany. If there is interference comming in from one of the connected repeaters,  sending a # does not disconnect immediately, but waits for a gap in the incomming audio stream before actually executing the disconnect command. If the incomming audio stream is completely uninterruppted, the command might never be executed. Not shure, if there's a command timeout implemented..

I propose that this behaviour be changed to 'disconnect immediately, even if an audio stream is incomming'.

73s, Axel DG1BBQ

sm0svx, 2017-07-17:
When you say "linked repeater system", do you refer to EchoLink or something else?

pe1chl, 2017-07-18:
We have some patches for a "remote control panel", a webpage (usable from mobile phone) allowing pre-registered users to control the behaviour of the repeater:

- disconnecting Echolink users
- disabling remote receivers

Parts of the required changes in svxlink have been submitted and accepted in a modified way, now the remote control panel itself will have to be modified to be compatible with that.

DG1BBQ, 2017-07-18:
Hi Tobias,

We have one computer on a fast Internet connection running svxlink in simplex logic. That's the 'server' all repeaters in the linked system connect to using echolink. Very simple and works really well. This way all repeaters are independent and have the option to disconnect from the system an talk localy. I added an external script to all involved repeater that handles autoconnect in place of svxlink's autoconnect. It takes care  that a repeater reconnects to the server after 4 minutes of local inactivity. It evaluates PublishSateEvent information to do this.
Also implemented  a bunch of administrative commands to allow me to disconnect a repeater or user in case of interference and also set a one time delay , so it wont reconnect after the standard time, but after a set  number off minutes. disabling/enabling autoconnect is also handled.
 I hope this clears all questions. Try to connect to one of the repeaters in the system (i.e. DB0BHV-R) and see for yourself.

73s, Axel.

sm0svx, 2017-07-18:
Thanks for the description. I now have a better picture of what you are doing. When the SvxReflector is released that would probably be a better solution for you, unless there are non-SvxLink based EchoLink clients involved. However, that does not change the need for being able to immediately disconnect a continuously transmitting EchoLink station. I would say that this is a bug.

DG1BBQ, 2017-07-18:
Non SvxLink:
Many OMs use their smartphones or laptops to work their home repeater while they are on holliday or are in foreign countries from all over the planet. That's an important ability we want to retain.

I'd like to thank you for the tireless work you are doing  for the benefit of all of us!