sm0svx/svxlink #298: Switchable TX-Audio Filtering Enhancement

DG1BBQ, 2017-07-18:
Using a transmitter that is designed only for 25kHz channel spacing on a 12.5kHz channel would require a hardware splatter filter with the upper -3db point set so that the modulation parameters are in line with regulations.
I suggest adding the capability of selecting between 25 and 12.5kHz tx-audio filtering.
Maybe also for 10kHz spacing on 10m. Also maybe rx-audio independentely? Hmmm...
Doing this in software most probably requires less effort than designing and building a hardware filter, except for the person implementing it ;)

I personally think anything below 20kHz sucks. Unfortunately the regulatory authority thinks differently.

sm0svx, 2017-07-19:
Adjusting a transceiver for operation is more than about splatter filtering. I argue that you should be able to properly adjust the volume settings so that everything work just fine with a 25kHz transceiver on a 12.5kHz channel using the current software. Just lower the output volume sufficiently and the current splatter filter will do its job. If possible, use the "devcal" utility to set the correct maximum deviation.

Now you might argue that the volume is too low, that there is not enough "punch" in the modulation. That really is another thing that I'd like to fix with SvxLink, the ability to shape the audio with something like AGC, compression, equalizer or just a separate manual audio gain setting. Getting the perfect audio is a complex matter but even with the existing features you should be able to adjust a repeater for balance between input and output audio so that the output audio sound just like the input audio. Then you at least have not destroyed anything.

I'd like to do some more work on audio level adjustment so that you get the same result for each calibration. The ear should not be involved in this calibration, just measurement. The clipper and splatter filter probably need to be tweaked but I don't think that it in the end need to be adjustable.

If you anyway want to try out another setting for the clipper, modify the [clipper creation code](https://github.com/sm0svx/svxlink/blob/master/src/svxlink/trx/LocalTx.cpp#L431). Something like:
```
AudioClipper *clipper = new AudioClipper(0.8);
```
1.0 is the default.

pe1chl, 2017-07-21:
We require the clipper in one repeater (23cm) where the transmitter is a re-purposed "packet radio link transmitter" that has no audio handling whatsoever.   I submitted a small patch that we now use to have a settable audioclipper limit.
Ideally the lowpass cutoff frequency would be selectable for the different bandwidths.
I think the architecture of svxlink w.r.t. audio processing is already very advanced, and many things including the above could be enabled by allowing (optional of course, with default settings same as the current hardwired values) control of the parameters of:
- compressor
- clipper level
- lowpass filter
For a really advanced clipper you would of course want a (software emulation of a) RF clipper, maybe even something like an OptiMod, but that would be room for further enhancement.

DF2HS, 2023-03-17:
Hi there

Some time ago Adi DL1HRC did some programming for me... Svxlink is using audio filters internally and he wrote something which allowed me to use these filters in the svxlink configuration to handle it flexible.

Sadly this code doesn´t work with the current version anymore.

It´s not just about limiting the audio level, it´s more about limiting the frequency range which is going to be used.

I did several tests... even the same type of a radio can behave different with the same settings and at the moment I would have to hardcode the filters to try out and compile every time again.

It would be so nice if these low pass filter with high Q could be integrated into svx configuration.

73
Sandy DF2HS