sm0svx/svxlink #302: Echolink Conference

ve3gob, 2017-08-05:
I am interested in experimenting.
Is it possible to use svxlink server as a radio-less conference server? I have scoured the docs and found no reference to conferencing, or a pseudo radio.
I have been experimenting with others such as thebridge1.09 and like it very much.
Allstar is fine but echolink integration is very limited. (No echolink texts can be received or sent, there is no sysop control via Text box for radio less install)
I would love to experiment with svxlink

pe1chl, 2017-08-15:
Sure this is possible.  You can define a radio on a soundcard that you don't use, but it is
also possible to use alsa:null as the sound device and configure a repeater without any
local radio at all.

ve3gob, 2017-08-15:
I will try the null idea. I am in canada and my servers are all over the
usa. No physical access.

----------- SIGNED -----------
              -Doug-
VE3GOB and VA3TDG
Sudbury, Ontario,
Canada

sm3sgp, 2019-09-10:
Please use the groups.io mailing list for further discussions.

closed 2019-09-10 by sm3sgp