sm0svx/svxlink #304: Too many messages...

iw0red, 2017-08-19:
Hi,
in the log file there are so many messages (continuous) like these:
----
256 samples of 512 written to the squelch detctor
256 samples of 512 written to the squelch detctor
256 samples of 512 written to the squelch detctor
256 samples of 512 written to the squelch detctor
256 samples of 512 written to the squelch detctor
256 samples of 512 written to the squelch detctor
256 samples of 512 written to the squelch detctor
256 samples of 512 written to the squelch detctor
[...]
can I decrease that?
73 Marco IW0RED

pe1chl, 2017-08-24:
What are your settings for the squelch parameters?  (SQL_*)
Try to reset them to defaults and make small changes until you reproduce this....

iw0red, 2017-08-25:
Hi,
The SQL_* : 
SQL_DET=CTCSS
SQL_START_DELAY=0
SQL_DELAY=0
SQL_TAIL_ELIM=250
----
I don't have the ctcss decoder on the radio what can I change?
73 Marco IW0RED

sm0svx, 2017-08-28:
What CTCSS_ parameters do you have setup?  Try to comment them out to get default behavior.

iw0red, 2017-08-30:
These:
CTCSS_FQ=88.5
CTCSS_SNR_OFFSET=-4.82
I can't comment them, if I do that the Rx1 doesn't work.
73 Marco IW0RED

iw0red, 2017-09-06:
News?
IW0RED

sm0svx, 2017-09-09:
Sorry for not coming back to you. Did you solve the problem?

iw0red, 2017-09-09:
Hi, that's fine no problem.
I didn't solve the problem, I have just disabled the log writer. :-)
73 Marco IW0RED

sm0svx, 2017-09-21:
OK. That's really not a solution since half of the audio will get lost and the squelch will not work as well as it should. I therefore open this issue up again and mark it as a bug.

iw0red, 2017-09-21:
I think so... 
73 Marco IW0RED

sm0svx, 2017-10-21:
Should be fixed now

closed 2017-10-21 by sm0svx