sm0svx/svxlink #307: Time Announce drops leading zeros in minutes portion

Dloranger, 2017-09-10:
When using the "*" dtmf code to force announce, the time is announced but when the minutes are less than 10, the leading zero is dropped, so the time is announced for example as "7 7 pm".  This is akward and confusing behavior.  The leading zero in the minutes should be announced so it would instead be "707 pm"

sm0svx, 2017-09-21:
It is the default behavior to include the leading zero. Have you changed anything or are you running a non-standard SvxLink distribution or sound pack?

Dloranger, 2017-09-21:
I just looked into the code more closely and I believe you have a typo in the locale.tcl file

set minute "O$minute";

that looks like the letter "O" not the number "0",

not sure your function call to play 2 digit number will know what to do with a letter instead of a number.  I haven't made it that far into the code yet...

Dloranger, 2017-09-21:
also to answer your question I am using a sound package from Rich Neese that is based on the Laura voice as I am developing modules and need to be able to expand the sounds and want to have the voices all match.

Dloranger, 2017-09-21:
I believe I may have found the source of the issue,  I am using a different sounds package as heather is no longer available for sounds development <I am developing modules so need to expand the sounds files>

In the sounds package I am using I do not have a O.wav, I have an o.wav (lower case).  I will clone this file to have both variants and try to reproduce the bug when I am at a station where I have audio available.

sm0svx, 2017-09-21:
The Heather voice IS still available. Just log in to your [Acapela Box](https://acapela-box.com/) account and make sure you pay for credits. The Heather voice should be there.

sm0svx, 2017-10-21:
Hmmm... It seems after all that Acapela have replaced Heather with the Sharon voice. Heather is still available on my account though. For how long I don't know. Anyway, this issue is solved I guess so I'm closing it.

closed 2017-10-21 by sm0svx