sm0svx/svxlink #311: Compiling Problems on Raspberry Pi Debian Stretch

dl1esk, 2017-09-23:
After upgrading my Raspbian Jessie to Raspbian Stretch i get the following Compiling Errors...
Any Ideas ?!

47%] Building CXX object locationinfo/CMakeFiles/locationinfo.dir/AprsTcpClient.cpp.o
In file included from /home/svxlink/src/locationinfo/AprsTcpClient.cpp:58:0:
/home/svxlink/src/build/include/common.h:146:6: warning: ‘bool SvxLink::setValueFromString(ValueType&, const string&) [with ValueType = std::__cxx11::basic_string<char>]’ defined but not used [-Wunused-function]
 bool setValueFromString(std::string &val, const std::string &str)
      ^~~~~~~~~~~~~~~~~~

53%] Building CXX object svxlink/trx/CMakeFiles/trx.dir/LocalTx.cpp.o
In file included from /home/svxlink/src/svxlink/trx/LocalTx.cpp:70:0:
/home/svxlink/src/build/include/common.h:146:6: warning: ‘bool SvxLink::setValueFromString(ValueType&, const string&) [with ValueType = std::__cxx11::basic_string<char>]’ defined but not used [-Wunused-function]
 bool setValueFromString(std::string &val, const std::string &str)
      ^~~~~~~~~~~~~~~~~~

dl1esk, 2017-09-23:
also...

 59%] Building CXX object svxlink/trx/CMakeFiles/trx.dir/SquelchEvDev.cpp.o
In file included from /home/svxlink/src/svxlink/trx/SquelchEvDev.cpp:52:0:
/home/svxlink/src/build/include/common.h:146:6: warning: ‘bool SvxLink::setValueFromString(ValueType&, const string&) [with ValueType = std::__cxx11::basic_string<char>]’ defined but not used [-Wunused-function]
 bool setValueFromString(std::string &val, const std::string &str)

sm0svx, 2017-10-01:
Yes, I know about that one. It's a "feature" of newer compiler versions. It is harmless but of course should be fixed. If it annoys you, you can disable the printout using:
```
cmake -DCMAKE_CXX_FLAGS=-Wno-unused-function ..
make
```

sm0svx, 2017-10-21:
Fixed in latest master

closed 2017-10-21 by sm0svx