sm0svx/svxlink #312: Bugfix in AsyncAudioDeviceAlsa/AsyncAudioIO  doesnt work on raspberry pi 2

dl1esk, 2017-09-23:
This Bugfix makes trouble on Raspberry Pi2 after update no sound and Repeater is still up then

Axel-DG1BBQ, 2017-09-24:
I have the same problem. When starting SvxLink on the Raspberry Pi3b will key up the TX, but will not key down again.

Axel-DG1BBQ, 2017-09-24:
Here's a quick fix for the Raspberry3b (should work elsewhere too)
At leatst on my sytems the working libraries
libasyncaudio.so.1.4.0.99.2 
libasynccore.so.1.4.0.99.2 
libasynccpp .so.1.4.0.99.2 
were not deleted.

go to /usr/lib/arm-linux-gnuabihf:
edit the link libasyncaudio.so.1.4 to point to libasyncaudio.so.1.4.0.99.2
edit the link libasynccore.so.1.4 to point to libasynccore.so.1.4.0.99.2
edit the link libasynccpp.so.1.4 to point to libasyncaudio.so.1.4.0.99.2 

delete the faulty libraries:
libasyncaudio.so.1.4.0.99.3
libasynccore.so.1.4.0.99.3
libasyncaudio.so.1.4.0.99.2

Shoud work right away. if not, just restart svxlink.

73s, Axel DG1BBQ

sp2ong, 2017-09-26:
I have updated my svxlink with latest sources and all working OK and I have the following version of libraries:
wrz 22 10:45 libasyncaudio.so.1.4 -> libasyncaudio.so.1.4.99.4-svxreflector
wrz 22 10:25 libasyncaudio.so.1.4.99.4-svxreflector
wrz 22 10:45 libasynccore.so.1.4 -> libasynccore.so.1.4.99.4-svxreflector
 wrz 22 10:23 libasynccore.so.1.4.99.4-svxreflector
 wrz 22 10:45 libasynccpp.so.1.4 -> libasynccpp.so.1.4.99.4-svxreflector
wrz 22 10:23 libasynccpp.so.1.4.99.4-svxreflector

73 Walek SP2ONG

dl1esk, 2017-09-26:
Hi Walek,

where do u get that Libary number.....?!
latest sources have:

 2 src/versions
@@ -5,7 +5,7 @@ QTEL=1.2.2.99.1
 LIBECHOLIB=1.3.2.99.0
 
 # Version for the Async library
-LIBASYNC=1.4.0.99.2
+LIBASYNC=1.4.0.99.3 !!
 
 # SvxLink versions
 SVXLINK=1.5.99.9

and they don`t work on the Raspberry Pi 2/3

sp2ong, 2017-09-26:
Hi,

I forgot to add that I have use svxlink with support svxreflector links. I have the similar problem on "RPI Zero W" after update to the new version svxlink but I have removed all libraries svxlink from /usr/lib/... and again run make install all works. My way to compile svxlink with support svxreflector links in following:

git clone https://github.com/sm0svx/svxlink.git
cd svxlink
git pull
cd src/svxlink
git checkout svxreflector
cd ../
mkdir build
cd build
cmake -DUSE_QT=OFF -DCMAKE_INSTALL_PREFIX=/usr -DSYSCONF_INSTALL_DIR=/etc  -DLOCAL_STATE_DIR=/var -DCMAKE_BUILD_TYPE=Release ..
make
(before make install I have remove all old libraries svxlink)
make install
ldconfig

73 Waldek SP2ONG

dl1esk, 2017-09-26:
the old libaries are not the problem because the new libaries will be copied over there and correctly linked.
but that versions doesn`t work.
if u change the link to the old libaries everything is fine

sp2ong, 2017-09-26:
Ok 
But from my hardware with "Raspberry Pi Zero W" with last update svxlink  all working. I have used on RPI Zero W Raspbina Stretch
It may be it is problem with RPI 2/3 

73 Waldek

sp2ong, 2017-09-27:
Hi

I confirm the problem with latest update svxlink sources. I have the update on other my echolink computer with HP T5735 and when I have start svxlink PTT is On and never OFF without any sounds. When I have switched back to the previous version libasyncaudio v1.4.99.3-svxreflector instead v1.4.99.4-svxreflector all working OK

73 Waldek

sp2ong, 2017-09-29:
I have yet another problem with the latest svxlink update. When I use alsa equalizer for [Tx1] and when the first time PTT is turned on is no audio and PTT does not turn off. I must manually stop svxlink. The svxlink without update works correctly with alsa equalizer.

n7ipb, 2017-09-29:
I'll also confirm this.  
Fresh RPi raspian install with v1.5.99.12-svxreflector.  Once keyed the  repeater never unkeys and the log shows squelch not closing.
Even running devcal -r alone without ever having started svxlink fails.

sm0svx, 2017-10-01:
That's unfortunate. It works fine here on my Cubieboard and my laptop. I'll try to fire up my Pi2 to see if I can reproduce the problem.

A tip; when running software directly from a version controlled repository, learn a couple of commands to revert back to the previous version in case of trouble. Mixing old libs with newer software (by removing the new libs) is not to be recommended even it worked in this case. What you should do is something like this:
* git checkout ORIG_HEAD
* Clean out old libraries, build and install as usual. 
* When you want to go back to using the latest version, run: "git checkout master" (or whatever branch you want to be on).
There are more ways to revert to a specific version of the code but this is the easiest one that will revert your source code tree to the state you were before doing a "git pull" or other "dangerous" operation.

sp2ong, 2017-10-01:
Hi Tobias,

Thank you for advice I have tried using git checkout ORGI_HEAD from master but without success:

cd svxlink/
git checkout master
Already on 'master'
Your branch is up-to-date with 'origin/master'.
git checkout ORIG_HEAD
error: pathspec 'ORIG_HEAD' did not match any file(s) known to git.

73 Waldek SP2ONG

sp2ong, 2017-10-01:
I have use svxreflector branch and I would like to have stayed with sources from GIT on SHA
7f2a063153538bb064f75c15eb70495ef8f29c96 from Commits on Jul 28, 2017

sm0svx, 2017-10-01:
Strange that ORIG_HEAD did not work. You can just do
```
git checkout 7f2a063153538bb064f75c15eb70495ef8f29c96
```
to get a specific revision.

sm0svx, 2017-10-01:
After many hours I finally got SvxLink running on the RPi2. I had problems getting audio input to work until I remembered that the RPi have problems with delivering enough current on the USB bus for some devices. Everything started working when I connected a USB hub.

Unfortunately everything seem to work on my Rpi. No hanging squelches or tx:es.

Can someone post a rx config that produce the error on their system? Especially interesting that @n7ipb got the problem with devcal. Maybe you can post your rx config?

sp2ong, 2017-10-01:
Hi Tobias,

I have the problem with HP Terminal T5735 with latest update svxlink , my Rx1 config:
[Rx1]
TYPE=Local
AUDIO_DEV=alsa:plughw:1
AUDIO_CHANNEL=0
SQL_DET=SIGLEV
SQL_START_DELAY=1000
SQL_DELAY=50
SQL_HANGTIME=150
SQL_TAIL_ELIM=200
SIGLEV_DET=NOISE
SIGLEV_SLOPE=-54.82
SIGLEV_OFFSET=364.87
SIGLEV_OPEN_THRESH=25
SIGLEV_CLOSE_THRESH=10
DEEMPHASIS=0
PEAK_METER=0
DTMF_DEC_TYPE=INTERNAL
DTMF_MUTING=1
DTMF_HANGTIME=40

But all we have problem with Tx1 because after start Tx1 PTT ON and never OFF and no audio
My Tx1:

[Tx1]
TYPE=Local
#AUDIO_DEV=alsa:plughw:1
AUDIO_DEV=alsa:equal
AUDIO_CHANNEL=0
PTT_TYPE=SerialPin
PTT_PORT=/dev/ttyS0
PTT_PIN=RTS
PTT_HANGTIME=500
TIMEOUT=1800
TX_DELAY=500
PREEMPHASIS=0
DTMF_TONE_LENGTH=100
DTMF_TONE_SPACING=50
DTMF_DIGIT_PWR=-18

I have use for Tx1 pass audio via alsamixer equalizer which work with svxlink before last update

sm0svx, 2017-10-01:
Yes, TX config is good to have as well. Thanks!

Do you mean that the TX goes on without the squelch ever being open?  I thought the TX hang because the squelch never closes?

sp2ong, 2017-10-01:
Hi,
This problem appears for example 
I have to start svxlink and when first time TX ON never TX OFF and never transmit any audio. The TX ON start for example when audio incoming from svxreflector link  or after start svxlink and I have PTT ON in my radio and svxlink typical send roger beep to confirm received signal but svxlink TX ON and not send roger beep and never OFF, I must manual stop svxlink

sp2ong, 2017-10-01:
I think we need more feedback  from others who try use latest updates

sm0svx, 2017-10-01:
Ok. Sounds like we should focus on the TX.

n7ipb, 2017-10-01:
I just did a fresh install with the minimum necessary to build and run svxlink.

New SD card,  2017-09-07-raspbian-stretch.img

I installed the usual support  then checked out the latest svxlink.
git clone https://github.com/sm0svx/svxlink.git
and checked out svxreflector

Configured with: cmake -DCMAKE_INSTALL_PREFIX=/usr -DSYSCONF_INSTALL_DIR=/etc -DLOCAL_STATE_DIR=/var -DUSE_OSS=NO -DUSE_QT=NO -DWITH_SYSTEMD=YES ..

Then build and install.

The attached gpio and svxlink config files were installed and I ran devcal after first doing svxlink_gpio_up

For the receiver: 
./devcal /etc/svxlink/svxlink.conf -r RxRepeater
The normal messages were output but not the line that normally shows the deviation and it takes a ctrl-c to get back a prompt.

Then for TX:
./devcal /etc/svxlink/svxlink.conf -t TxRepeater
The normal header shows then when you press 'T' the transmitter comes on but it doesn't go off when you press 'T' again and there is no audio transmitted.  The only way out is to do a ctrl-c.

To make it even more confusing, if I start svxlink everything works.

Oct 01 2017 22:32:37.242: RxRepeater: The squelch is OPEN (82.3206)
Oct 01 2017 22:32:37.874: RxRepeater: The squelch is CLOSED (3.05542)
Oct 01 2017 22:32:43.669: TxRepeater: Turning the transmitter OFF

Yesterday it wasn't working with a more complex setup so I'll go back to that and see if there are any additional clues.

Ken - N7IPB

[gpio.conf.txt](https://github.com/sm0svx/svxlink/files/1347550/gpio.conf.txt)
[svxlink.conf.txt](https://github.com/sm0svx/svxlink/files/1347575/svxlink.conf.txt)

n7ipb, 2017-10-02:
Ok,  I went back to using the svxlink.conf file that wasn't working yesterday,  and indeed it also fails to run today.   The squelch opens,  the repeater transmits...and hangs until I stop svxlink.

I was able to narrow it down to one part of the configuration.

On all my systems I have darkice running to stream the audio so I have an alsa loopback device loaded with 'sudo modprobe snd_aloop'

I have the attached asound.conf file in /etc 

Add the mtx.txt file to the end of the svxlink.conf from my previous message and change the RepeaterLogic TX to RepeaterMultiTx

When RepeaterMultiTx contains TxStream svxlink will lock up.  Remove it and it works fine.

Prior to the latest Async changes this worked fine.

Ken
[asound.conf.txt](https://github.com/sm0svx/svxlink/files/1347651/asound.conf.txt)
[mtx.txt](https://github.com/sm0svx/svxlink/files/1347652/mtx.txt)

dl1esk, 2017-10-02:
I also have a loopback Device in my RX Config, i use it once a week to send Amateur Radio News..
My Rx Section:

[Rx1]
TYPE=Local
AUDIO_DEV=alsa:hw:1
GPIO_SQL_PIN=gpio3

[Tx1]
TYPE=Local
AUDIO_DEV=alsa:plug:plugequal
AUDIO_FILTER=BpBu4/50-4000
PTT_PIN=gpio17
PTT_TYPE=GPIO
GPIO_PTT_PIN=gpio17

I have the same that no Audio then and when Squelch opens once it stays open and TX doesnt switch off anymore.

I will do a Test agn today without the Equalizer.
But for good Audio i need that Equalizer...

dl1esk, 2017-10-02:
i narrowed down the problem....
if u use AUDIO_DEV=alsa:plug:plugequal
in ur TX to make the Audio Quality better the problem occurs
if u open the repeater  the squelch will be open and doesnt close and tx will be on all the time
if u use svxlink without equalizer everything is fine
Tobias do u have an idea whats going wrong there ?!

sp2ong, 2017-10-05:
Hi Tobias

I have to try to find how to get version sources from git for branch svxreflector before update AysnAudio

When I have run following

git checkout svxreflector 

I switch to svxreflector
but now I would like to have in branch svxreflector source with SHA 7f2a063153538bb064f75c15eb70495ef8f29c96

and run

git checkout 7f2a063153538bb064f75c15eb70495ef8f29c96 

it is OK look sources are before last update but I don't have svxreflector source in svxlink tree

sm0svx, 2017-10-21:
Thanks for the detailed reports! Sorry I haven't been able to deal with the problem until now. I'll try to set up a config with a loopback device to see if I can trigger the problem.

@sp2ong, hmmm, yes... I guess it's not that simple. Try "git log --first-parent svxreflector". You probably want commit f071bca6b6c2de6ad3f31f9db84de969913f80a6.

sm0svx, 2017-10-21:
I managed to reproduce the problem on my laptop using the config that @n7ipb provided. Now I just have to find the bug...

sm0svx, 2017-10-21:
Bug fixed in latest master and svxreflector branches. Please tell me if it's working now in your setup.

sp2ong, 2017-10-22:
Hi Tobias,

I have to get the fresh code from github and switch to svxreflector and replace on all my local nodes to the new version and reload new svxreflector.
All node where I have used alsa loop and alsa equalizer now working without audio problems.
Thank you for fix problem.
73 Waldek

n7ipb, 2017-10-22:
The latest code seems to have fixed the bug introduced by the  AsyncAudioIO bug fixes.   I no longer see the problem with PTT.

Unfortunately I had hoped that issue #209 would be solved by the AsyncAudioIO changes but I'm still seeing the assertion when using CTCSS.

sm0svx, 2017-10-22:
That problem is probably not connected directly to any audio I/O issue but rather have something to do with code closer to the squelch/CTCSS handling.

This issue seem to be solved though so I'm closing it.

closed 2017-10-22 by sm0svx