sm0svx/svxlink #313: Possible to interface with rpitx?

yavolfoxxen, 2017-09-23:
I've been wondering if it would be possible to interface this with another project (https://github.com/F5OEO/rpitx) to be able to use the on-board GPIO pin for local TX (given proper filtering, of course).

yavolfoxxen, 2017-09-29:
Ionno, maybe this should be a feature request?

sm0svx, 2017-10-01:
Cool idea. Have you tested it on ham radio frequencies?  Any measurements when it comes to RF compliance?  I can't imagine that it generates a clean signal. With external filtering and power amplification maybe it can be useful?  Good for demos as well.

F1TZO, 2017-10-01:
Hi, 
Yes, svxlink work fine with rpitx. I did a configuartion 1 year ago, you can do it ;-)
You have to make a script to define how commute RX/TX, I did it with the GPIO, connect 2 GPIO both and detect SQL and read TX value. With this, you active rpitx to TX with the script, and it's work fine.

yavolfoxxen, 2017-10-02:
I've played with it on the 2m band, filtering is needed, lots of harmonics peaks because square waves. Will interfere with anyone in range who happen to listen on peaks when not using a filter (or one with insufficent attenuation in spurrious range)

Intended to use for short range use (around the house with a small ht), and a rtl-sdr for rx.

F1TZO that is good to hear. As far as i can understand this one would need to convert the audio stream first to a .wav file, then to a .ft file with pifm to be able to transmit with rpitx. This chain would not be real-time i fear.

ae7ov, 2018-10-02:
Very interested in this. Just got rpitx
running on rpi3b+ and breifly tested 
in "vco" mode. 
Also have ft8call working on same
rpi and would like to use rpitx to make self contained transceiver but
dont really have skills to do so..now.
Gary