sm0svx/svxlink #315: METARinfo play during echolink module active

sp2ong, 2017-11-02:
Then is active echolink module we cannot listen to any info from METAR information module.
METAR is a very nice module which delivers information about weather conditions from defined airports.
In many cases, echolink module occupation most of the time and it will be nice to add subcommand to echolink module for play METAR information if are defined and loaded to svxlink. 
According to information  https://github.com/sm0svx/svxlink/wiki/ModuleEchoLink we can add for example subcommand 70# to play METAR info for defined airport as STARTDEFAULT in config file
When we send DTMF 71# echolink module will be playing METAR info for the first airport defined in AIRPORTS= in config file, when we send DTMF 72# we will listen info for second airport if defined in AIRPORTS= .....

73 Waldek SP2ONG

f5nlg, 2017-11-02:
> svxlink/wiki/ModuleEchoLink we can add for example subcommand 70# to play

Hello
When Echolink module IS présent you have to type *51# and you got meteo
anouncement it works ...

I run svxreflector with propagation monitor without Time Out and i do the
same , *51# 1st airport *52# 2nd airport etc ...

73" f5nlg

sp2ong, 2017-11-02:
F5NLG 
Thank you for the info, I have missed information in doc svxlink that I can use * additionally.
Yes, you right and works

73 Waldek SP2ONG

closed 2017-11-02 by sp2ong