sm0svx/svxlink #316: WARNING: Illegal WAV file header. ChunkSize should be Subchunk2Size plus 36

sp2ong, 2017-11-03:
I have use tools audacity to create wav files with local information and time to time I have to see in svxlink log

WARNING: Illegal WAV file header. ChunkSize should be Subchunk2Size plus 36

What is reason generate this message by svxlink ??.  I have played this wav file via sox player without problem, but svxlink don't want play this file

73 Waldek

sm0svx, 2017-11-04:
Your file failed because it contained RIFF/WAV subchunks (LIST and id3) that SvxLink did not know about. That is to be considered a bug since a properly implemented RIFF parser should just ignore unknown subchunks. That have been fixed in latest master.

closed 2017-11-04 by sm0svx

sp2ong, 2017-11-04:
Tobias 
I have use svxreflector and this update is not available for this branch. Cna you update this changes in svxreflector branch ???
Thank you

Waldek

sm0svx, 2017-11-04:
Done

sp2ong, 2017-11-04:
I have to try compile in branch svxreflectro but:

[ 72%] Building CXX object svxlink/svxlink/CMakeFiles/svxlink.dir/MsgHandler.cpp.o
/home/sp2ong/src/svxlink/src/svxlink/svxlink/MsgHandler.cpp: In member function ‘virtual bool WavFileQueueItem::initialize()’:
/home/sp2ong/src/svxlink/src/svxlink/svxlink/MsgHandler.cpp:754:22: error: ‘errno’ was not declared in this scope
          << strerror(errno) << endl;
                      ^
/home/sp2ong/src/svxlink/src/svxlink/svxlink/MsgHandler.cpp:790:24: error: ‘errno’ was not declared in this scope
            << strerror(errno) << endl;
                        ^
/home/sp2ong/src/svxlink/src/svxlink/svxlink/MsgHandler.cpp:817:26: error: ‘errno’ was not declared in this scope
              << strerror(errno) << endl;
                          ^
/home/sp2ong/src/svxlink/src/svxlink/svxlink/MsgHandler.cpp:843:28: error: ‘errno’ was not declared in this scope
                << strerror(errno) << endl;
                            ^
/home/sp2ong/src/svxlink/src/svxlink/svxlink/MsgHandler.cpp:895:28: error: ‘errno’ was not declared in this scope
                << strerror(errno) << endl;
                            ^
/home/sp2ong/src/svxlink/src/svxlink/svxlink/MsgHandler.cpp:930:43: error: ‘errno’ was not declared in this scope
          << filename << "\":" << strerror(errno) << endl;
                                           ^
/home/sp2ong/src/svxlink/src/svxlink/svxlink/MsgHandler.cpp: In member function ‘virtual int WavFileQueueItem::readSamples(float*, int)’:
/home/sp2ong/src/svxlink/src/svxlink/svxlink/MsgHandler.cpp:954:44: error: ‘errno’ was not declared in this scope
          << filename << "\": " << strerror(errno) << endl;
                                            ^
/home/sp2ong/src/svxlink/src/svxlink/svxlink/MsgHandler.cpp: In member function ‘virtual void WavFileQueueItem::unreadSamples(int)’:
/home/sp2ong/src/svxlink/src/svxlink/svxlink/MsgHandler.cpp:978:44: error: ‘errno’ was not declared in this scope
          << filename << "\": " << strerror(errno) << endl;
                                            ^
svxlink/svxlink/CMakeFiles/svxlink.dir/build.make:60: polecenia dla obiektu 'svxlink/svxlink/CMakeFiles/svxlink.dir/MsgHandler.cpp.o' nie powiodły się
make[2]: *** [svxlink/svxlink/CMakeFiles/svxlink.dir/MsgHandler.cpp.o] Błąd 1
CMakeFiles/Makefile2:1327: polecenia dla obiektu 'svxlink/svxlink/CMakeFiles/svxlink.dir/all' nie powiodły się
make[1]: *** [svxlink/svxlink/CMakeFiles/svxlink.dir/all] Błąd 2
Makefile:137: polecenia dla obiektu 'all' nie powiodły się
make: *** [all] Błąd 2

sp2ong, 2017-11-04:
You can see exactly the same problem with compile in:
https://travis-ci.org/sm0svx/svxlink/jobs/297287362

sm0svx, 2017-11-04:
Fixed. Try again.

sp2ong, 2017-11-04:
Ok, Thank you. Now I have success with compile on Orange Pi Zero and HP 5730. I will try to use wav file to play via svxlink which generate the problem.

sp2ong, 2017-11-04:
The wav file which previously generated problem now was played without problem
Thank you for fixed problem with wav filles

73 Waldek