sm0svx/svxlink #317: Kernel Crash with actual SvxLink Version

dl1esk, 2017-11-07:
Hi Tobias,

with the new Version on an Raspberry Pi 3 i become an Kernel Problem and the whole Raspi hangs up:
i was in the Console and became the following error Message:

Message from syslogd@DB0CA at Nov  7 22:58:00 ...
 kernel:[ 2277.323701] Internal error: Oops: 17 [#1] SMP ARM

Message from syslogd@DB0CA at Nov  7 22:58:00 ...
 kernel:[ 2277.345574] Process svxlink (pid: 875, stack limit = 0xb66bc210)

Message from syslogd@DB0CA at Nov  7 22:58:00 ...
 kernel:[ 2277.346891] Stack: (0xb66bde80 to 0xb66be000)

Message from syslogd@DB0CA at Nov  7 22:58:00 ...
 kernel:[ 2277.348218] de80: b65b3f00 00000006 b66bdef4 b66bde98 8017f230 802725                                   70 60000113 80b7ad40

Message from syslogd@DB0CA at Nov  7 22:58:00 ...
 kernel:[ 2277.350892] dea0: 80c79814 80c0f5c0 b65b36c0 ba360f80 b920e070 80c723                                   47 80b7f080 80c0312c

Message from syslogd@DB0CA at Nov  7 22:58:00 ...
 kernel:[ 2277.353601] dec0: 80c031d0 80c0fb40 b66bdef4 80c020a4 00000009 b66bc0                                   38 00000100 b9c05180

Message from syslogd@DB0CA at Nov  7 22:58:00 ...
 kernel:[ 2277.356309] dee0: 00000024 00000200 b66bdf5c b66bdef8 8010169c 8017ec                                   d8 801011e8 b9c88100

Message from syslogd@DB0CA at Nov  7 22:58:00 ...
 kernel:[ 2277.359048] df00: 00000001 00400040 80c02d00 0003043f 00000008 80c7b9                                   80 80c726d8 80c0312c

Message from syslogd@DB0CA at Nov  7 22:58:00 ...
 kernel:[ 2277.361927] df20: 80b793b8 80c02080 b66bdef8 0000000a 80178484 80b7ce                                   c0 00000000 00000000

Message from syslogd@DB0CA at Nov  7 22:58:00 ...
 kernel:[ 2277.364944] df40: 00000001 b9c05180 00000000 ffffffff b66bdf6c b66bdf                                   60 80122cf8 8010151c

Message from syslogd@DB0CA at Nov  7 22:58:00 ...
 kernel:[ 2277.368087] df60: b66bdf94 b66bdf70 8017371c 80122bf8 b66bdfb0 80c046                                   70 40000010 ffffffff

Message from syslogd@DB0CA at Nov  7 22:58:00 ...
 kernel:[ 2277.371383] df80: 10c5383d 10c5387d b66bdfac b66bdf98 8010150c 801736                                   b8 80c040a4 767cdba4

Message from syslogd@DB0CA at Nov  7 22:58:00 ...
 kernel:[ 2277.374752] dfa0: 00000000 b66bdfb0 8071aa0c 80101470 015b6c48 7ef8fd                                   18 00000000 7ef8fe04

Message from syslogd@DB0CA at Nov  7 22:58:00 ...
 kernel:[ 2277.378202] dfc0: 00000007 015b6c48 00000003 00000008 00000065 000000                                   00 ffffffff 00000004

Message from syslogd@DB0CA at Nov  7 22:58:00 ...
 kernel:[ 2277.381699] dfe0: 00000008 7ef8fc44 00000009 767cdba4 40000010 ffffff                                   ff e1a00007 0a00008c

i had that Error twice..last week one time and today...
any Ideas ?!

dl1esk, 2017-11-08:
i will try to narrow it a bit down because i did 2 Changes...first i changed Hardware from RPI2 to RPI3 and Yesterday when i had this Error twice i tried using dwc_otg.speed=1 ( forcing USB 1.1 ) from the raspi and then this Error happened. On the Raspi 2 i didnt had this failure in this combination.
First i will try to go back to the previous Version without the Alsa fixes to see if thats the Issue or not

sp2ong, 2017-11-08:
I had described problem kernel Ops .. on Rasberry Pi Zero board with Debian v9 and svxlink before alas fix I have tried with Raspbian Jessie but the similar problem was with kernel Ops . I thought that this problem is connected to hardware RPI Zero and for this reason, we are changing the board to Orange Pi Zero and all working without problems. I have tried at RPI Zero with dwc_otg.speed but without success.

dl1esk, 2017-11-08:
on the Raspi 2, i could use dwc_otg.speed=1 but then sometimes the audio was dropped for a few seconds and then the Repeater worked fine again...
Now i changed to Raspi 3 and had that problem... Without dwc_otg.speed=1 i had that problem once...
Maybe Tobias has an Idea how to narrow down the problem then...

sm0svx, 2017-11-08:
Sorry but I have no idea. Sounds like a hardware problem or kernel/driver bug. Your best bet is probably to contact the RPi community. No bug in SvxLink should be able to crash the kernel so something is definitely wrong with OS or hardware.

dl1esk, 2017-11-09:
I already posted it in the rpi community, when i get news i post it here

dl1esk, 2017-11-09:
Hi Tobias,

i could narrow down the problem ab bit better now...

Out of memory: Kill process 545 (svxlink) score 893 or sacrifice child
Nov  9 15:30:15 DB0CA kernel: [ 2091.606963] Killed process 545 (svxlink) total-vm:904472kB, anon-rss:872512kB, file-rss:0kB, shmem-rss:0kB

more Info`s later

sm0svx, 2017-11-09:
Hmmm, that's a bit strange I must agree. Can you monitor SvxLink using "top" to see if it's leaking memory?  In top, press "M" to sort on memory usage.

dl1esk, 2017-11-10:
Hi Tobias,

i got a fix for that Problem...
i had to use combination dwc_otg.fiq_fsm_enable=0 & dwc_otg.speed=1 in cmdline.txt together then no crash happens.
@sp2ong u can try that too if u want, should work on Raspi Zero too...
i try to get the reason why this happens...

dl1esk, 2017-11-10:
@sm0svx i observed that sometimes the first 5-10 Secs of an passage doesnt come through ... still no sound then from the station that speaks.. After that time the rest of the passage comes through...
then everything resumes as normal... Maybe the open command get lost to open the speaker channel ?!
is there any chance to recheck  that the open command got received by the HW ?! Because it happens only at the start of an passage not in the middle or at the end...
i know that the output gets opened/closed everytime when a passage starts/stops

sp2ong, 2017-11-10:
Hi Sebastian,

Ok I will back again to Rasberry Pi Zero and try to suggested by you options in cmdline.txt

73 Waldek

gicardous, 2017-11-10:
Maybe is related with JITTER_BUFFER_DELAY parameter. On one of my nodes everything worked ok on version v1.5.99.13 with JITTER_BUFFER_DELAY=1000
Switching to latest release and the audio won't come out of the node, I mean no PTT, no sound.
On the receiving side was ok, other nodes would hear me speaking.
Commenting out that line and everything went back to normal.
But we need that parameter because we use several mobile (3g/4g) connected systems and that adjustable jitter helped a lot.
I did not experienced kernel crash or other errors. System is armbian on a nanopi.
We also have nearly 40 active nodes waiting to be switched to svxreflector but we still miss the rx disable feature implemented in voter,

73 Catalin yo7gqz

dl1esk, 2017-11-10:
@gicardous that was with previous versions too and using dwc_ogt.speed=1
So thats no fault from the latest version

sm0svx, 2017-11-14:
@gicardous: If there is a problem with JITTER_BUFFER_DELAY please report it in another issue.

@dl1esk: Audio getting lost also seem like another issue. Report it as a new issue if it's not related to this one. There already are a lot of error checks in the Alsa code but no error seem to be reported by the Alsa driver in your case.

sm0svx, 2017-11-15:
@dl1esk: The 10 seconds of audio getting lost should be solved now in latest master.

dl1esk, 2017-11-16:
@sm0svx i will check that..has the svxreflector something to do with the normal Repeater operation?

sm0svx, 2017-11-16:
@dl1esk: Everything in SvxLink is connected through the audio paths so an error in one end of SvxLink can have unexpected effects on other parts. The repeater functions should not have been affected by the "10 second" problem though.

sp2ong, 2017-12-01:
Hi,

I have back again to Raspberry Pi Zero with latest svxlink sources and use dwc_otg.fiq_fsm_enable=0 & dwc_otg.speed=1 in cmdline.txt.
Now svxlink on RPI Zero W working without kernel crash one week and it looks that was solved the problem.

73 Waldek

closed 2017-12-02 by sm0svx