sm0svx/svxlink #318: svxlink symbol lookup error since merge of svxreflector into svxlink

Axel-DG1BBQ, 2017-11-11:
Installed the current svxlink with merged svxreflector. When trying to run svxlink, I now get the following error:
**svxlink: symbol lookup error: svxlink: undefined symbol: _ZTIN5Async9TcpClientE** and svxlink quits.

I have tried removing all files and libs  that belong to svxlink and doing
"git clone https://github.com/sm0svx/svxlink.git" and recompiling. Compiles without error, but I still get the same result.
Any suggestions as to what might be the problem? Has anyone encountered this error?
73s, Axel

f5vmr, 2017-11-11:
Yes - Me too........ New install of svxreflector on the server, a recompile and install over the old version, and now the identical error
Chris F5VMR

Axel-DG1BBQ, 2017-11-11:
Thanks Chris for the confirmation!
Unfortunately more troubleshooting for poor Tobias.

Axel-DG1BBQ, 2017-11-11:
Installed the current svxlink with merged svxreflector. When trying to run svxlink, I now get the following error:
svxlink: symbol lookup error: svxlink: undefined symbol: _ZTIN5Async9TcpClientE and svxlink quits.

I have tried removing all files and libs that belong to svxlink and doing
"git clone https://github.com/sm0svx/svxlink.git" and recompiling. Compiles without error, but I still get the same result.
Any suggestions as to what might be the problem? Has anyone encountered this error?
73s, Axel

Dammit! closed it by mistake.

dl1hrc, 2017-11-12:
I can not confirm this error. I've upgraded six stations with different
hardware/software combinations (opensuse, debian, armbian, i386, amd,
arm7) without any problems.

vy 73 Adi / DL1HRC

lechner, 2017-11-12:
Are your executables finding the updated shared libraries? Have you tried running `ldconfig`?

WU8K

dl1hrc, 2017-11-12:
yes, and double check that you don't have old libs under /usr/lib/... /usr/lib64 /usr/linux-xxx-lib or something like this. The installation is sometimes a bit strange.
vy 73s de Adi / DL1HRC

Axel-DG1BBQ, 2017-11-12:
I've since installed it on another SBC and svxlink works. So I checked everything against the working system. Ifount that the incudes and libs were not replaced by the new version as were'nt the modules. I manually fixed that using the files from the build folder and  yes I had rum ldconfig before.
So, I dont know what happended durring installation. I'm going tinstall the currend build over the existing one, just to see, what happens. BTW the SBC where the Problem occured ist a orangepiPC1.
If Chris F5VMR is also running a similar model, then the Problem might be limited to orangePis.
Before anyone asks... I made myself an installation script that i have used repeatedly to install on the orangepi and others systems without any trouble. 

Since it's not an actual svxlink error, I'm closing this Issue.

73s, Axel DG1BBQ

closed 2017-11-12 by Axel-DG1BBQ

f5vmr, 2017-11-12:
Aha - I installed on a PC, but I am now looking at the install for the lib files. Good thinking Batman! I'll watch for it on the Raspberries too when I do the upgrade. Thanks