sm0svx/svxlink #319: how to react to GPIO status changes

Dloranger, 2017-11-13:
I am looking for a way to create code that would detect a GPIO event and be able to report an alert over the air.  Essentially long term it would become a station monitor that could report a number of issues to alert of a station issue without a human having to request the information actively.

Something like an intrusion detector, fire alarm, fuel level alert, etc.

My understanding of modules precludes this as an option as modules cannot be concurrently operating as I understand it, so if a station monitor module is running, it would prevent other modules from being used as I understand the limitations around the modules.  the monitor module would have to be stopped, the new module loaded and used, and then the monitor reactivated.

Am I missing something that would create this functionality?  

I would be okay with some precanned messages (or preferably read from a file perhaps??) that respond to a given input gpio, these could be loaded as generic variables in the config file perhaps.

StationMonitor1=FireAlarm
StationMonitor1Pin=GPIO100
StationMonitor1MSG=/usr/share/FireAlarm.txt
StationMonitor2=DoorSensor
StationMonitor2Pin=GPIO101
StationMonitor2MSG=/usr/share/DoorSensor.txt

sm0svx, 2017-11-14:
A module can emit notifications even though it's not currently active so it should fulfill your needs. What an inactive module cannot do is transmit and receive audio "normal" audio.

Dloranger, 2017-11-22:
thanks for the feedback,  by chance do you have a framework I could work outward from for such functionality. no sense reinventing the wheel if you have already got something, thanks

sm0svx, 2017-11-27:
Have look in the [TCL module subdirectory](https://github.com/sm0svx/svxlink/tree/master/src/svxlink/modules/tcl) for an example of how to write a module in TCL or in [the template directory](https://github.com/sm0svx/svxlink/tree/master/src/svxlink/modules/template) if you prefer writing a module in C++. I have not worked with GPIO but if you ask on the svxlink-devel mailing list you may find someone that have.

Dloranger, 2017-12-07:
I have gotten some experience with the modules in general, and I plan initially to just read the value files of the gpio on a long interval, like 2x per second to see if they have changed state.  not optimal, but good enough to develop some code.

The issue I see so far is the modules all appear to be waiting for a command to be received, while I want one that just launches and runs a continuous loop.  I don't find any examples of how to do this.

Ideally I don't want this to be something the user interacts with at all, just a watchdog that monitors for events.

I envision this entire code block being something like this

while {1} {
   read gpio file value
   if value != oldvalue
       if value =1
           playMsg "ALARM"
}

Dloranger, 2017-12-07:
I have some functional code, but everything I have tried causes the module loader to hang as the system has to be put into a while(1) loop.

Can you advise how to go about having the module run a polling process without hanging the setup process?

###############################################################################
#	SVXlink Site Status Module
#	Coded by Dan Loranger (KG7PAR)
#  
#	This module enables the user to configure sensors to monitor the health and
#   wellbeing of a remote site.  The module runs in the background and on a
#   regular interval (typically a few times a second) checks the inputs and will
#   announce over the air, any configured messages as appropriate to alert the
#   site manager/monitors that an event has occurred.
#
###############################################################################

#
# This is the namespace in which all functions and variables below will exist.
# The name must match the configuration variable "NAME" in the
# [ModuleTcl] section in the configuration file. The name may be changed
# but it must be changed in both places.
#
namespace eval SiteStatus {
	# Check if this module is loaded in the current logic core
	#
	if {![info exists CFG_ID]} {
		return;
	}

	#
	# Extract the module name from the current namespace
	#
	set module_name [namespace tail [namespace current]]
	
	
	# A convenience function for printing out information prefixed by the
	# module name
	#
	#   msg - The message to print
	#
	proc printInfo {msg} {
		variable module_name
		puts "$module_name: $msg"
	}
	
	proc main {} {
		variable CFG_DIGITAL_0
		set gpioPath "/sys/class/gpio/gpio"
		# capture the initial values on the sensors and enable them if the settings are 
		# all defined (validity of the sensor is not enforced)
		if {[info exists CFG_DIGITAL_0]} {	
			printInfo "Digital Sensor 0 is enabled"
			set CURRENT_STATE_0 [exec cat $gpioPath$CFG_DIGITAL_0/value]
			set DIGITAL_0 ENABLED
			printInfo "Digital Sensor 0 is enabled"
		} else {
			set DIGITAL_0 DISABLED
			printInfo "Digital Sensor 0 is disabled"
		}
		
		while {1} {
			printInfo "While Loop Running"
			# 2 samples per second should be frequent enough
			after 500 
			# Digital Sensors
			if {$DIGITAL_0 == "ENABLED"} {
				# Read the updated state of the sensor and compare against the old state
				# alerts should only go out when the sensor changes state
				set NEW_STATE_0 [exec cat $gpioPath$SITESENSOR_DIGIAL_0/value]
				if {$CURRENT_STATE_0 != $NEW_STATE_0} {
					set CURRENT_STATE_0 $NEW_STATE_0
					switch $SITESENSOR_DIGIAL_0_TYPE {
						DOORSENSOR{DOORSENSOR_STATUS{0 $NEW_STATE_0}}
						#TEMPSENSOR{TEMPSENSOR_STATUS{0 NEW_STATE_0}}
						default {printInfo "SENSOR 0 is of unknown type"}
					}
				}
			}
		}
	}

	#basic function to announce the door sensor changing status
	proc DOORSENSOR_STATUS {sensor value} {
		playMsg site_status "door"
		playNumber $sensor
		if {$value == 1} {
			playMsg site_status "open"
		} else {
			playMsg site_status "closed"
		}
	}

	#
	# Executed when this module is being deactivated.
	#
	proc deactivateCleanup {} {
		printInfo "Module deactivated"
	}
	

# end of namespace
main
}

#
# This file has not been truncated
#

sm0svx, 2017-12-07:
You must never block/loop/wait in a module since that will block the whole application. You should use a timer for polling. Have a look at the propagation monitor [here](https://github.com/sm0svx/svxlink/blob/master/src/svxlink/modules/propagation_monitor/ModulePropagationMonitor.tcl#L283-L284) and [here](https://github.com/sm0svx/svxlink/blob/master/src/svxlink/modules/propagation_monitor/ModulePropagationMonitor.tcl#L269-L272).

Dloranger, 2017-12-07:
thanks, I will see if I can make this work with the suggested codes

closed 2017-12-18 by Dloranger