sm0svx/svxlink #320: JITTER_BUFFER_DELAY value

gicardous, 2017-11-14:
Hi,
We used to set JITTER_BUFFER_DELAY=1000 in version v1.5.99.13
in version v1.5.99.15
JITTER_BUFFER_DELAY=1000
will not start the transmitter.
lower values will not work either. Commenting it out seems to solve the problem.
73! catalin yo7gqz

sm0svx, 2017-11-15:
Fixed a couple of bugs in the ReflectorLogic but I'm not sure that have anything to do with this problem. I cannot reproduce it. Do you still see the JITTER_BUFFER_DELAY problem in latest master?

gicardous, 2017-11-16:
On my node the problem is gone. Will update the other nodes and I'll be back with an answer.
Thank you.

gicardous, 2017-11-16:
Hi Tobias,
I can confirm, JITTER_BUFFER_DELAY problem gone.
Thanks again.

closed 2017-11-16 by sm0svx