sm0svx/svxlink #321: Enable/disable receiver in svxreflector

gicardous, 2017-11-15:
Hi,
we would like to have this feature presented in svxlink also implemented in svxreflector.
The reason is that often there are more people listening than a temporary disruptive signal which can occur on one receiver, affecting all network. Blocking with firewall would cut all the traffic from that node.
Using rx disable would let the traffic from the network on tx side of that node.

Thank you,
Catalin yo7gqz

sm0svx, 2017-11-15:
Can you be a bit more specific? Do you mean such a feature should be implemented on the server side, in the SvxReflector software or on the SvxLink side?  How would it be controlled?

sq4bja, 2017-11-15:
Hi.
        
    In Poland we have an FM-LINK network based on
          SVXLink + SVXReflector. It consists of 13 repeaters. The ability to block its elements is very useful.
        Quickly turn off the intruder, do not turn off the whole
          network.
        In our case, we used
          Domoticz software to control the network block. This is a very
          useful software that integrates perfectly with the GPIO port
          in raspberry PI. We see all the states of each
          repeater, see which receiver it receives, and so on. To block
          reception, we used the channel change function on the Motorola
          MTR2000. GPIO
              high-level switching to the receiver channel, preventing
              interference. This
          is a very practical solution that can be implemented in
          SVXLink, but using DTMF is not practical. Control capability should be implemented via the
          internet. This
          gives you 100% administrator identification, full
          authorization and speed. I attach to the post photos of our panels
          with full visualization and control of the network.
    Tom SQ4BJA
    
    W dniu 2017-11-15 o 06:33, gicardous
      pisze:
    
    
      Hi,
        we would like to have this feature presented in svxlink also
        implemented in svxreflector.
        The reason is that often there are more people listening than a
        temporary disruptive signal which can occur on one receiver,
        affecting all network. Blocking with firewall would cut all the
        traffic from that node.
        Using rx disable would let the traffic from the network on tx
        side of that node.
      Thank you,
        Catalin yo7gqz
      —
        thread.

sm0svx, 2017-11-15:
Please be careful when using email to reply to an issue. Do not include quotes of previous posts. That just makes the conversation flow look messy. Attachments are not supported. You should be able to attach files in the web UI though.

sq4bja, 2017-11-15:
Ok, Thanks Tobias, and sorry for the confusion.
Below screenshots of dashboards of our network (i`m using WWW interface)

[image]
[image]
[image]

gicardous, 2017-11-16:
Hi Tobias,
Yes, such a feature should be implemented on the server side, in the SvxReflector software.
Te be more specific, with this command issued, the SvxReflector should 'absorb' rather then reflect traffic coming from that specific node but transmit to that node traffic from the other nodes.
This feature is already implemented in SvxLink but you have to put the receiver through a voter:

[LocalVoter]
TYPE=Voter
RECEIVERS=Rx1
COMMAND_PTY=/dev/shm/voter_ctrl

so issuing 'echo "DISABLE Rx1" > /dev/shm/voter_ctrl' instantly disables that receiver and 'echo "ENABLE Rx1" > /dev/shm/voter_ctrl' enables it. 
I'm aware that there is no voter in SvxReflector but this is just an example of how we do this now in SvxLink
In SvxReflector any method would be ok as long as it can be issued through a web page.

All the best!
Catalin yo7gqz

sm0svx, 2017-11-16:
@sq4bja: Looks very nice :-)

@gicardous: Some form of remote control interface should be added to both SvxLink and SvxReflector. It would probably be easiest to start with SvxReflector since that is a much simpler application. Howerver, I have some other things that I want to do first before starting to add new features to SvxReflector so it may take some time until such a feature is implemented.

pe1chl, 2017-11-23:
The method of interfacing between external control interfaces and svxlink using a PTY is actually working well.  We have created several (3) different control interfaces for the ENABLE/DISABLE of a receiver in the Voter that way.
The latest one required control from a different system, so a form of remote procedure call that works over the network would have been convenient, but that was easily solved using a small daemon that listens on a port and forwards the received data to the PTY.
(with appropriate firewalling of course)
Extending that type of PTY interface into svxreflector could be the best way.

riku2015, 2021-05-24:
any progress on this? im also interested of this function. even a simplest pty command will be ok. best will be timed, like mute for 1hour if there is static or something.