sm0svx/svxlink #322: Simulcast with svxlink

iw0red, 2017-11-15:
Hi Tobias, we are going to try to realize a simulcast network composed of five svxlink repeaters, but if it's work fine many more. :-)
Actually we have one svxlink repeater whith five receivers with voting system and we could configure another one for make some test. What can you suggest to use? svxreflector? remotetrx? I have no idea what's the best choise.
To mantein the trasmitter oscillators to the frequency specification the GPS unit is used to provide a constant 10MHz signal reference. 
73 Marco IW0RED.

sm0svx, 2017-11-15:
Please use the svxlink-devel mailing list to ask questions. The issue tracker is for reporting bugs and requesting new features. If you by Simulcast mean co-channeling/coversity like the dutch do, neither SvxReflector nor (unpatched) RemoteTrx will work for this I think.

closed 2017-11-15 by sm0svx

iw0red, 2017-11-16:
Hi, this is a new features request. It's for my bad english maybe... 
73 Marco IW0RED

sm0svx, 2017-11-16:
What is needed is what is discussed in issue #85, proper audio synchronization. Right now I think that is mostly implemented outside of SvxLink.

iw0red, 2017-11-17:
Fine, thanks.
73 Marco IW0RED