sm0svx/svxlink #323: svxlink hang when using automatic disconnect from svxreflector and station is talking

dl7ata, 2017-11-18:
Hi Tobias,
we use cronjob connects and disconnects for netlink connections into our svxreflector system.
So for example on saturday at 14:02 the netlink will be disconnect by
02 14 * * 6 /bin/echo "*930#" > /tmp/svx_pty.RemoteLogic
command. When at that certain time somebody is talking on netlink, on both of my logics - 2m and 70cm - the PTT is permanently on and runs after several minutes into timeout. After that, I have to restart svxlink. Would be great, If you can solve that issue.
Thanks, Frank

sm0svx, 2017-11-19:
That's not good. Thanks for reporting. I have not tested much enabling/disabling the link. I'll have a look at it.

sm0svx, 2017-12-04:
Found a bug in a pretty central audio component, the Async::AudioSelector. Fixed in latest Git master.

closed 2017-12-04 by sm0svx

dl7ata, 2017-12-05:
Thanks for your great support!

Have a nice x-mast time Tobias.

 

Best

Frank

 

Closed #323 <https://github.com/sm0svx/svxlink/issues/323> .