sm0svx/svxlink #326: Alsa loop and CPU used by svxlink

sp2ong, 2017-11-23:
I have used Alsa loop with defined logic in svxlink to send bulletins (wav file) which are transmitted on Air on the local node and to others nodes connected via svxreflector.

I noticed that when I use Alsa loop in configuration of  svxlink, the  "top" tool show me that CPU used by svxlink is  about 25% 
When I switch off Alsa loop from configuration of svxlink the CPU used by svxlink drop to about 3 to 7 %

My question is it normal % of CPU used by svxlink when we use ALSA loop or not ? 

73 Waldek

pe1chl, 2017-11-23:
We use Alsa loop as output device and do not observe this issue.  Maybe it happens only on input.
For sending bulletins we use remotetrx, maybe you can use it as a workaround?
(in our system the bulletins are sent from remote systems so remotetrx is the only solution but of course you can run remotetrx on the same system and connect via 127.0.0.1)

sp2ong, 2017-11-24:
Hi Rob,

Thanks for the suggestion.  I have use latest version svxlink

 I have configured remotetrx with Alsa loop RX and connect logic remoteTRX in svxlink via 127.0.01 to remotetrx. But now I have 👍 
svxlink about 3-5% CPU
remotetrx about 17-20 %CPU

My config remotetrx.conf:
[GLOBAL]
TIMESTAMP_FORMAT="%c"
CARD_SAMPLE_RATE=48000
TRXS=NetUplinkTrx

[NetUplinkTrx]
TYPE=Net
RX=RxLoop
TX=NONE
MUTE_RX_ON_TX=1
LISTEN_PORT=5300
FALLBACK_REPEATER=0
AUTH_KEY="authkey"

[RxLoop]
TYPE=Local
AUDIO_DEV=alsa:plughw:2
AUDIO_CHANNEL=0
SQL_DET=VOX
SQL_START_DELAY=0
SQL_DELAY=10
SQL_TAIL_ELIM=0
SQL_HANGTIME=1300
VOX_FILTER_DEPTH=20
VOX_THRESH=10
DEEMPHASIS=0
DTMF_MUTING=1

sm0svx, 2017-11-27:
I'm not sure what's causing this. Can you post the loop configuration as well?

sp2ong, 2017-11-28:
Hi Tobias,

The loop configuration:

pcm_slave.hw_loopback {
   pcm "hw:Loopback,1,2"
   channels 2
   format RAW
   rate 16000
}
pcm.plug_loopback {
   type plug
   slave hw_loopback
     ttable {
     0.0 = 1
     0.1 = 1
   }
}

73 Waldek

ur3qjw, 2024-09-04:
SvxLink v1.8.0 I notice the same thing.
High load on the processor when using Alsa loop. Or when using UDP socket.
Debian 11 Linux 5.10.0-31-amd64
ASYNC_AUDIO_ALSA_ZEROFILL=1
ASYNC_AUDIO_UDP_ZEROFILL=1
Alsa plugins and svxlink modules are not used.

Immediately after launch, and during the first PTT, the load does not exceed 1.3-2%.
When the transmission is complete, the transmitter is switched off.
The load on the processor increases. First to 15%, and then to 43%.
During subsequent transmissions, the load is reduced to 2%.
In standby mode, increases to 43%.

[image]

This can be confirmed by simply combining two logics:
[GLOBAL]
LOGICS=SimplexLogic,SimplexLogic-2

[SimplexLogic]
TYPE=Simplex
RX=NONE
TX=Tx1
#MODULES=ModuleEchoLink

[SimplexLogic-2]
TYPE=Simplex
RX=Rx1
TX=NONE
#MODULES=ModuleEchoLink

[Rx1]
TYPE=Local
#RX_ID=?
AUDIO_DEV=alsa:hw:2,0,0
or
AUDIO_DEV=udp:127.0.0.1:4810
AUDIO_CHANNEL=0
SQL_DET=VOX

[Tx1]
TYPE=Local
TX_ID=T
AUDIO_DEV=alsa:hw:2,1,0
or
AUDIO_DEV=udp:127.0.0.1:4810
AUDIO_CHANNEL=0

closed 2025-03-13 by sp2ong