sm0svx/svxlink #328: Decoder options

dl1hrc, 2017-11-24:
Hi Tobias,
I would like to talk about the handling of en-/decoder options in the Async framework when configuring Speex or Opus. The way is to create an en-/decoder instance and then send some parameters to it.
In my opinion this could be made some troubles since you can not use the instance as long as not all parameters has sent to it and the program flow can not know how many parameters are still pending.
Do you think that it could be changed that way that the full parameter set is handed over when the instance will be created? The setOption() function may remain in the source, it's a good way to change the parameters on-the-fly but the initializing should be made with the full parameter set. 
What do you think?

vy 73s de Adi / DL1HRC

sm0svx, 2017-11-27:
I'm not sure I understand when this would be a problem. The setOption function is used in two ways in SvxLink. One use is when [reading codec configuration options from a configuration file](https://github.com/sm0svx/svxlink/blob/master/src/svxlink/trx/NetRx.cpp#L200-L213) at initialization. Then you can read all options and set all options on the codec one by one before going any further.

The second use can be found in [RemoteTrx](https://github.com/sm0svx/svxlink/blob/master/src/svxlink/remotetrx/NetUplink.cpp#L580-L586) when the MsgRxAudioCodecSelect message is received. All options are contained in the same network message so all options can be applied at once.

dl1hrc, 2017-11-28:
Yes, but in both cases you have created an instance first and **later** you set options on it. For the most cases it maybe ok but for my Dmr stuff I see some problems. E.g. I have to create an instance that will open a serial port for communication, a lot of failures may happen here. You can not set a default port (like /dev/ttyUSB0). So the flow looks like:

1 inst = new Deocoder();
2 inst->setOption("PORT","/dev/ttyS0");
3 inst->setOption("SPEED","9600");
...
with 2 or 3 the setOption() may fail but I get not back this information in the caller instance. Ok, at the end of initializing process I can call up another method like inst->isInitComplete(); but then I have an instance that I can not use any more. I've no deeper knowing with concurrent processes but it could create more problems if another part of the application wants to use the instance before the initializing process has been finished. 
According to my non-authoritative opinion it would be a bit better this way:
Decoder *inst = new Decoder(map<parameter, value>);
if (inst == 0) 
{  
  cout << "error\n";
  delete inst;
  return false;
}

sm0svx, 2017-11-28:
Of course, Adi. I didn't think of the case where you are talking to a hardware codec. I now understand the problem.

Your suggestion with the map is good but I would prefer adding a new function, setOptions(map<string,string>), instead of modifying all the constructors plus the create method. Then you do not have to modify all other codecs. You just provide a generic AudioDecoder::setOptions method that call the setOption function for all values in the config map. In your codec implementation you reimplement the setOptions method so that you can apply all parameters in one "transaction".

I understand that using a separate function may leave your codec in a half initialized state but you could handle this by just throwing away incoming samples until the codec is initialized. It would be good to handle an error state gracefully anyway if someone pull the hardware out temporarily. Also, most of the time you would probably be able to call the setOptions function directly after creating the object.

Sounds good to you?

sm0svx, 2017-11-28:
I'm not sure what would be the best strategy for error handling. Maybe it's ok for the codec initialization to fail and we hope for the problem to fix itself, for example when someone plug the correct hardware in. A printout to inform the user of the problem is nice though.

I used this strategy when talking to the RTL dongle. SvxLink may be started and the dongle can be plugged in later or even pulled while SvxLink is running. It makes everything pretty solid. Like this:
```
SimplexLogic2: Event handler script successfully loaded.
*** WARNING: No supported RTL devices found
*** WARNING: No supported RTL devices found
<plugging the dongle in>
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
WbRx1: Changed state to READY
        Using device      : [0] Realtek RTL2838UHIDIR SN:00000001
        Tuner type        : R820T
        Valid tuner gains : 0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48 49.6 
```

dl1hrc, 2017-11-28:
hm....the dmr stuff is (must be) implemented as a multiton and called up from rx and tx instances, makes it not easier I think.

sm0svx, 2017-11-29:
I don't understand why it's such a big difference to make two calls (constructor/setOptions) instead of one when you create a coder/decoder. It's essentially what you do anyway right now only that setOptions is automatically called from the constructor.

The drawback with doing complicated stuff in the constructor, like initializing connections to external hardware, is that such things may fail and failure is harder to handle in the constructor than in a separate initialize method. That is why I often split creation/initialization. Stuff that cannot fail or only fail in extreme cases (e.g. out of memory) can be put into the constructor but things that are expected to fail (e.g. USB device not found, configuration error etc) is put in a separate method.

In this case it would be logical to handle configuration errors as fatal, since they would never correct themselves, while handling a lost connection to an USB device as temporary failure. The latter may fix itself if the user plugs the USB device in again.

It would be good if you could merge the latest master into your DmrLogic branch. The pull request looks quite messy right now.

KG5RKI, 2018-02-12:
sm0svx  If you are interested in a dmr/p25/c4fm codec and bridge, please contact me via email.  KG5RKI