sm0svx/svxlink #330: remoteTRX and Dynamic IP adress

sp2ong, 2017-11-29:
Hi

I have use remoteTRX in svxlink.conf
I have used dynamic IP on remoteTRX and in svxlink.conf I have setup in following

TYPE=Net
HOST=sp2abs.ddns.net

I have a problem when remoteTRX change IP and svxlink try to use only IP address which resolves first time. I must restart svxlink to reconnect to new IP address remoteTRX. 

Maybe better will when svxlink will resolve the name to IP address every time when lost connection ???

73 Waldek

sm0svx, 2017-11-29:
Yes, that would be better. I thought I had changed this but obviously not.

sp2ong, 2017-12-15:
Does anyone know how to temporarily fix an issue with a changing IP address on remoteTRX via any script in crontab ... ????

sm0svx, 2017-12-18:
The method most people use I think is to set up a VPN tunnel between the sites using for example OpenVPN. Then you can use a static address for SvxLink/RemoteTrx within the tunnel.

sm0svx, 2017-12-18:
Latest master now does a name lookup on each reconnect so hopefully this will solve your problem. Of course the DNS have to update fast enough and time to live have to be sufficiently short for uninterrupted operation.

closed 2017-12-18 by sm0svx

sp2ong, 2017-12-19:
Hi Tobias,

Many thanks for fix this problem. I have checked and works very well now and I don't need to create VPN tunnel

73 Waldek