sm0svx/svxlink #331: Raspberry Pi Raspbian Jessie Install

HEWscsi, 2017-11-30:
I was looking forward to have svxlink operational on my Pi. When I followed the instructions in the Install.adoc I had a hiccough on cmake.
-- Reading versions file...
-- Checking for module 'sigc++-2.0'
--   Found sigc++-2.0, version 2.10.0
-- Found SIGC2: /usr/lib/arm-linux-gnueabihf/libsigc-2.0.so  
-- Found chown: /bin/chown
-- SvxLink user = svxlink
-- SvxLink group = daemon
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found RT: /usr/lib/arm-linux-gnueabihf/librt.so  
-- Could NOT find Speex (missing:  Speex_LIBRARY Speex_INCLUDE_DIR) 
--   Speex is an optional dependency. The build will complete
--   without it but support for the Speex audio codec will
--   be unavailable.
-- Could NOT find Opus (missing:  Opus_LIBRARY Opus_INCLUDE_DIR) 
--   Opus is an optional dependency. The build will complete
--   without it but support for the Opus audio codec will
--   be unavailable.
-- Checking for module 'alsa'
--   No package 'alsa' found
ALSA_INCLUDE_DIR=ALSA_INCLUDE_DIR-NOTFOUND
ALSA_LIBRARY=ALSA_LIBRARY-NOTFOUND
CMake Error at cmake/Modules/LibFindMacros.cmake:74 (message):
  Required library ALSA NOT FOUND.

  Install the library (dev version) and try again.  If the library is already
  installed, use ccmake to set the missing variables manually.
Call Stack (most recent call first):
  cmake/Modules/FindALSA.cmake:35 (libfind_process)
  async/audio/CMakeLists.txt:85 (find_package)

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GSM_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /home/pi/Downloads/svxlink-master/src/async/audio
   used as include directory in directory /home/pi/Downloads/svxlink-master/src/async/audio
   used as include directory in directory /home/pi/Downloads/svxlink-master/src/async/audio
   used as include directory in directory /home/pi/Downloads/svxlink-master/src/async/audio
   used as include directory in directory /home/pi/Downloads/svxlink-master/src/async/audio
   used as include directory in directory /home/pi/Downloads/svxlink-master/src/async/audio

-- Configuring incomplete, errors occurred!
See also "/home/pi/Downloads/svxlink-master/src/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/Downloads/svxlink-master/src/build/CMakeFiles/CMakeError.log".

Is there a possible future for svxlink on the pi?

Thanks,
dWB4SLI

lechner, 2017-11-30:
Hi, do you have the Debian package for version 15.11 in Raspian? Otherwise, you can build it with:

`dget https://packages.debian.org/source/stable/svxlink`
`cd {subdir}`
`debuild -b -uc`
`cd ..`
`dpkg -i {package name}.deb`

 Should you need more recent features, there are two additional options:

1. I can put together a Debian package for the current master branch.
2. You build from the GitHub source after installing the dependencies. Speex and Opus may be optional, but you probably need `libasound2-dev` for ALSA and `libgsm1-dev` for GSM.

Best regards,
WU8K

sm3sgp, 2017-12-08:
@HEWscsi Have you tried to install on Raspbian Stretch instead? In Raspbian Stretch svxlink is available as packages, and can be installed with apt commands:
sudo apt-get install svxlink-server

sm3sgp, 2018-02-10:
An updated instruction on how to install SvxLink on a Raspberry Pi with apt-get install is now available on the Wiki:
https://github.com/sm0svx/svxlink/wiki/InstallBinRaspbian

brettwi, 2018-02-11:
Is there a Wiki or web page that shows how to interface the Raspberry Pi in a repeater?

Thanks!

-Brett

https://github.com/sm0svx/svxlink/wiki/InstallBinRaspbian

Dloranger, 2018-02-11:
There are a few prebuilt options listed at Open repeater (mine and others)
but there are also guides being developed currently as well for DIY
hardware. They are being worked on actively but may not be officially
published yet. If you email the open repeater admin and tell him I pointed
you his way he will share the link(s).

Dan, kg7par

f5vmr, 2018-02-11:
Yesterday, with the latest master of Svxlink downloaded yesterday, and a pi b+ running the latest full version of Raspbian stretch, I installed a repeater. It worked from the offset with no issues. You can follow my progress on Facebook, svxlink Amateur Radio users, or on f5vmr.wordpress.com

sm3sgp, 2019-09-01:
This issue is a bit outdated. Several instructions on how to install svxlink on Raspbian exists on the svxlink wiki. Closing

closed 2019-09-01 by sm3sgp