sm0svx/svxlink #345: Initialization failed for module ModuleEchoLink?!

fahood007, 2018-01-01:
Hi...
Happy New year 
Thank sm0svx for the new update of svxlink.. it was interesting to try it with my raspi2 ...
although it was pointing to this issue " Initialization failed for module ModuleEchoLink"..

Pretty sure everything is fine from my side...

using your project for 2 years now

cheers!

f5nlg, 2018-01-01:
Hi to all

and HPNY 2018 !

testind on a server it works , no problems with ModuleEchoLink , i'm 
always doing .deb package with checkinstall so it's easy for me to 
remove previous version ...

before make the checkinstall , did you try ldconfig ? reboot ? remove 
old libraries before install ?

works also on OrangePi Zero.

73" to all , F5NLG

fahood007, 2018-01-01:
Hi..
Thanks on fast reply...
Well this is a fresh install on my PI2 with new SD card.. (Always keeping old as a backup)..

Have a gr8 day
73
Fahad
A92FQ

f5nlg, 2018-01-01:
Hi,

Sorry i do not use raspberry only orangePi based pico computers.

73"

sm0svx, 2018-01-02:
There should normally be an error printout preceding that error message telling you what the problem actually is.

fahood007, 2018-01-20:
thank you all for support

closed 2018-01-20 by fahood007