sm0svx/svxlink #346: adaptive jitter buffer algorithm

sp2ong, 2018-01-02:
Hi Tobias,

I am not sure what kind of algorithm jitter buffer is implemented in svxlink. I have found a nice description of benefits use adaptive jitter buffer:

http://www.linphone.org/news/52/26/Introduction-to-our-new-RTP-adaptive-jitter-buffer-algorithm.html

It maybe implement this algorithm to svxreflector will help in some cases with not good quality network

73 Waldek SP2ONG

sm0svx, 2018-01-02:
The jitter buffer used for reflector audio is very simple. On reception in the SvxLink node the audio playback will be delayed with the number of milliseconds set up using the JITTER_BUFFER_DELAY in the ReflectorLogic configuration. That delay will build up a buffer of audio to compensate for jitter.

The reflector itself does not do any jitter compensation. It will just distribute received audio packets without further delay.

This mean that all nodes will have to have their jitter buffer configured for the "worst" participating node. An adaptive jitter buffer delay would of course be nice but the importance of keeping the delay low is not as important on a two-way radio link as it is for a full duplex phone call. Some times it's just better to have a simple and predictive algorithm than having a complicated adaptive algorithm. Also, radio hams probably will not take much notice of an occasional buffer under-run audio drop out since they are used to all sorts of harsh RF conditions.

Maybe this subject will have to be revisited in the future if it turns out to be a big problem but for now the current implementation will have to do. The constant increase in quality of data networks will also make this less of a problem. We are running at least one of the nodes on a 3G (may be 4G now) connection and I have never heard any noticeable audio drop-outs despite using JITTER_BUFFER_DELAY=0 on my node.

What kind of connection are you using that cause jitter audio drop-out problems?

pe1chl, 2018-01-05:
The co-channel diversity program that I wrote for use with svxlink on our multisite repeater includes an adaptive delay method like that.  It quickly increases the delay when it sees packets that are too late, and slowly decreases it when all packets arrive in time.  A margin is maintained so not every packet later than the average causes an underrun.  Reading the above, I could consider adapting the margin to the variance seen in the packet delay times (when the times are very steady, the margin could be smaller).

sm0svx, 2018-01-06:
@pechl: Yes, that sounds like what I also would do if implementing dynamic jitter delay but it's low prio for me right now. One additional problem in the reflector is that the audio source changes all the time so it's harder to choose a good jitter buffer delay over time. One would probably have to maintain one delay parameter per each unique source (station/node). It's also hard to maintain a proper jitter buffer in SvxLink due to the construction of the audio pipe. There is simply no way at the moment to detect downstream under- or overflow.

n7ipb, 2018-01-16:
I'm currently trying to dial in the jitter buffer size needed for three sites.  I'm seeing as many as 300+ dropouts per five minute period on at least one of the three.  It's a fed by a commercial 5Ghz wireless service.  My other two SvxLink sites are on HamWan and they so far never see more than 5 dropouts per interval.   
I recently started graphing the dropouts detected by SvxLink so I can determine if it's time of day related or not. (thinking it might be congestion on the network)  
See the Inbound UDP packets graph here:
http://cacti.wetnet.net/cacti/graph_view.php?action=tree&node=tbranch-9&hgd=&hyper=true

Having a fixed buffer has had one noticable effect.  Users need to wait for the buffer on the linked system to empty before they can expect a response.  Since at least one of our systems is a hub for RF linked non-SvxLink repeaters I find that users on a SvxLink repeater can't break in because the back and forth exchange on the RF linked systems happens so rapidly.